SynthstromAudible/DelugeFirmware #1946: Pop/Click when triggering note in FM Mode

requiets, 2024-05-16:
### Please describe the problem:

Triggering notes in audition or iso keyboard creates a pop/clipping/click noise at the beginning of the sample. This is most noticeable when playing medium or harder velocity and is not heard when lightly pressing or playing with playing with portamento. Prevalent in FM synth mode but have also experienced in other modes, though not often.

### What is the expected behavior?

Note triggers pitch without additional pop/click noise.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

deluge-v1_2_0-nightly+2024_05_16-f636f84

### If possible provide the steps to reproduce the issue and upload additional media:

Selected FM Synth Mode, press pad in either ISO keyboard or audition with medium or higher velocity.

seangoodvibes, 2024-05-17:
DX7 FM synth or original FM synth?

requiets, 2024-05-17:
FM Synth. My mistake - I thought that was the same as DX7 FM.

sapphire-arches, 2024-05-19:
Is this different compared to prior firmware versions? if not, there's a few common causes: 
 - attack set to zero 
 - sample start not aligned to a zero crossing

m-m-adams, 2024-05-21:
Yeah, the fix is to set phase retrigger to 0 on the carriers. It's 5th pad from the bottom in the oscillator column  

We should probably do that by default in FM synths instead of having it off

m-m-adams, 2024-08-08:
Closing - clicks are already avoidable and they can be useful for sound design

closed 2024-08-08 by m-m-adams