SynthstromAudible/DelugeFirmware #1961: High CPU load / dropouts in community beta

MkUltra85, 2024-05-18:
### Please describe the problem:

The song was starting to flash the CPU high load indicator (play button) when the song contained:
7 samples
1 mono subtractive synth bass
1 subtractive synth pad (3-4 voices)
also some midi and cv clips.

I've added a few samples and an audio clip since then, and now it gets noticeable dropouts. Posting this in response to an exchange with Mark on Facebook. 
[24 May 17 Jam 2.zip](https://github.com/SynthstromAudible/DelugeFirmware/files/15366321/24.May.17.Jam.2.zip)

### What is the expected behavior?

I've expected more CPU headroom based on what others have said. I've been a bit surprised how quickly I hit the limit.

### Is there a relevant Pull request?

Not sure.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community, other

### What is the firmware name:

1.1.0-BETA-CC25FFF

### If possible provide the steps to reproduce the issue and upload additional media:

Just hitting play where it's saved will cause the CPU indicator to flash frequently for me.

sapphire-arches, 2024-05-21:
I think this is culling just because voice allocation is kinda slow, even with the static voice array. We'll probably improve this eventually, but at least to my ears it doesn't sound like anything "important" is getting culled -- just some note/sample tails and those are getting soft culled rather than hard culled. If there's a particular element that you think should be surviving but isn't, can you give the name of the track for that element?

m-m-adams, 2024-05-21:
This drops out occasionally on my stock SD card that I test with but plays without dropouts on my main one [link](https://www.amazon.ca/gp/product/B07H9DVLBB/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)

If you could check the format of your card that would be helpful - ideally it should be FAT32 (not exfat) with a cluster size of at least 32 KB

sapphire-arches, 2024-05-21:
It does not, and that might not help anyway -- we need an SD card with low random read latency which may or may not correspond to a higher overall speed. The M8 has similar requirements, so SD cards that are known to work well with the M8 should also be good for us.

If you have collect-all versions of songs that generate clicks or static we can take a look at those, that shouldn't happen.

The synth pad is "22-broken pads"? I didn't notice it dropping out, but one thing that might help is setting it to `HIGH` priority - the system will never cull higher priority voices before lower priority voices, so HIGH can be very useful for long pads or audio tracks that are very noticeable when they drop out.

Adding some instrumentation to audio rendering, it seems like the expensive bit is the RMR sample kit:

```
audio_engine.cpp:327: soft-culled 1 voice.  numSamples:  35. Voices left: 16. Audio clips left: 1
audio_engine.cpp:938:
audio_engine.cpp:942: 35:  soft cull
audio_engine.cpp:942: 43:  Start output render
audio_engine.cpp:942: 104:  complete: 002 SDS-5
audio_engine.cpp:942: 263:  complete: 001 DDD-1
audio_engine.cpp:942: 543:  complete: 22-Broken pads
audio_engine.cpp:942: 617:  complete: A Bss
audio_engine.cpp:942: 619:  complete:
audio_engine.cpp:942: 623:  complete:
audio_engine.cpp:942: 637:  complete: 000 TR-808
audio_engine.cpp:942: 639:  complete:
audio_engine.cpp:942: 663:  complete: AUDIO1
audio_engine.cpp:942: 1033:  complete: Mk N Rmr
audio_engine.cpp:942: 1119:  Reverb complete
audio_engine.cpp:942: 1121:  mastercomp start
audio_engine.cpp:942: 1145:  mastercomp end
audio_engine.cpp:945: 1149: end
audio_engine.cpp:494: direness:  13
```

This kit is pretty expensive because it has a large number (4-5) of stacked note-on events. Resampling that loop to an audio clip and muting the original allowed me to play the rest of the song without any culling at all.

MkUltra85, 2024-05-23:
I'm a slow human.

I ended up grabbing an SD card similar to the one @m-m-adams mentioned. I realized later that it is slightly different and might not perform the same: [this card](https://www.amazon.de/-/en/SanDisk-Transfer-QuickFlow-Technology-Temperature-Resistant/dp/B09X7FXHVJ/ref=sr_1_3?crid=BDUGCKYCG2KR&dib=eyJ2IjoiMSJ9._nqLdYHlrAMIJc4YNuvOf6XZmrokIj5Fk8wbxDSyD8Duon-X3ymE3q8YEXhebd6ldEXkuEB-gIVQ0NMyak7uRWUqXxk5mSOYyLuC9LtOqzBkrXw7biA-uKbAxyftW8WeWfs79Ra1cv9jJNwk6ZR9KmPh6IF_x0FGbadEWTZB6hTx4I4X3PWccOfAO0qQqaPsBMJK0f3njaEncfTFhOZU8R0O-sAidLkRVqDhZfaBASA.vcFvhhU24BSNNwJezb-CXc8vOs0HDa9qqwK1eIRv4LM&dib_tag=se&keywords=sandisk+extreme+pro+200&qid=1716496691&sprefix=sandisk+extreme+pro+200%2Caps%2C130&sr=8-3) The only difference I see is that it advertises 200mb/s instead of the 170mb/s on the other card, but it is also cheaper. I mention this because I tried the above track with this card, and I'm still getting dropouts. 

@sapphire-arches Yes, it was the "22-Broken Pads" synth. I'll use the HIGH priority going forward for sure.

I'm including a couple songs that I had issues with. I realize that they could be cleaned up a lot; the first one has a multisampled bass track and some kits that could obviously be resampled. Still, they create more clicks/pops/dropouts for me on the beta than on official. 

[Jan4.23 Within Groups.zip](https://github.com/SynthstromAudible/DelugeFirmware/files/15422615/Jan4.23.Within.Groups.zip)
If I go to the arranger view, and play this from 75:1:1, I get a loud static pop at 77:1:1 Many pops / clicks / dropouts throughout. I just verified that the static pop does not happen on official firmware.

[Jan10.23 Monte Carlo Simulation.zip](https://github.com/SynthstromAudible/DelugeFirmware/files/15422628/Jan10.23.Monte.Carlo.Simulation.zip)
This one is mostly FM instruments, but still generates a lot of clicks and pops for me. Another thing about this (and maybe I should make a new issue for this?) the FM BELL 1 synth causes a buzzy crackly distortion on the beta firmware that was not present on official. 

Thanks for the tips. I clearly need to make resampling a bigger part of my workflow.

sichtbeton, 2024-05-23:
How do these play on official?

MkUltra85, 2024-05-23:
@sichtbeton Not perfect but pretty close. I think the last one actually plays without any problems on official. The other, I might have one or two pops/clicks in a play-through. On beta, it happens much more frequently, and they are much more pronounced.

Also, greetings fellow Berliner.

sapphire-arches, 2024-05-24:
> The FM BELL 1 synth causes a buzzy crackly distortion on the beta firmware that was not present on official.

Even turning the level way down there's still a huge transient-like thing at the start of the sound, which seems odd

[image]

However, I loaded up C1.0.1 and it sounds the same on that release so either we broke this a really long time ago or the change was inherited from official 4.1.4. I can get it to not click at specific notes by messing with the retrigger phase, but that breaks when changing octaves. Please file a new bug for that problem, maybe there's some way we can fix it other than recommending ultra long envelopes.

That transient then gets in to the reverb buffers which  cause the distorted sound. I don't think we've changed the sound of the reverb, but maybe we should run some frequency and impulse sweeps against it to make sure...

m-m-adams, 2024-05-25:
> I ended up grabbing an SD card similar to the one @m-m-adams mentioned. I realized later that it is slightly different and might not perform the same: [this [...]

Do you know how it's formatted and its cluster size? 

I also think the more important speed for us is random read which might be different between them

MkUltra85, 2024-07-04:
> Do you know how it's formatted and its cluster size?
>
> I also think the more important speed for us is random read which might be different between them

Oops big lapse. I formatted it using a tool that didn't tell me (or allow me to change) the cluster size. Is there a tool you'd recommend for formatting?

m-m-adams, 2024-07-27:
https://github.com/SynthstromAudible/DelugeFirmware/wiki/SD-Cards

We've added info on SD cards - the 200mbps version of the card I linked is worse than the 170 version

I did some testing and the Kingston canvas go plus is significantly better than either, it's also the only card recommended for the dirty wave m8 so I think it's a safe bet. I get several more streaming samples with it than I did on the sandisk

closed 2024-07-27 by m-m-adams

MkUltra85, 2024-07-29:
@m-m-adams I really appreciate what you are all doing. 

Can I ask what software you use / recommend for formatting the cards? I haven't found a tool that checks all my boxes yet.