SynthstromAudible/DelugeFirmware #1966: Load song scrolling names stuck in loop

rpc9943, 2024-05-19:
### Please describe the problem:

Load song scrolling down from song files it suddenly starts looping, on my directory it stops at AE02 song file, under it is blank, and scrolling will loop from blank to AE01. Not sure why, but if i scroll opposite will go all the way until it hits same filenames.

### What is the expected behavior?

To scroll through entire song file list

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

other

### What is the firmware name:

1.1.0-BETA-EE0D48C

### If possible provide the steps to reproduce the issue and upload additional media:

Not sure how to reproduce on other sd card directories just reporting whats happening on my end

m-m-adams, 2024-05-19:
Can you share a listing of the directory contents?

macOS instructions:
1. Pop your sd card into your computer 
2. Open the sd card in finder
3. Cmd + click on the path at the bottom of the window and choose "open in terminal"
4. Type ls SONGS > songs.txt

Windows:
1. Open powershell
2. Type cd D:/ (or whatever letter the sd card is)
3. Type ls SONGS > songs.txt

Then upload songs.txt 

If that doesn't work a screenshot might help too

rpc9943, 2024-05-19:
[songs.txt](https://github.com/SynthstromAudible/DelugeFirmware/files/15369931/songs.txt)

Balistix55, 2024-05-20:
I had the same issue in november last year #692
It was caused by the corrupted file (files) on sd card. Try to find corrupted file. In my case it was a file with song and collected samples. You can try also another sd card.

m-m-adams, 2024-05-21:
I think it's the combo of 02Ae01.xml, 02Ae01 2.xml, Ae01.xml, and Ae01 2.xml with their collect all folders. The legacy browser probably sees those as the same name and is jumping between them 

I'm going to move this to future, it's niche and I don't think it should block the current release

m-m-adams, 2024-08-08:
Closing since we don't have a repro, it's probably the same bug as the other ones related to auto naming songs and I'm leaving the earliest one open

closed 2024-08-08 by m-m-adams