SynthstromAudible/DelugeFirmware #1972: suggestion: exiting pink mode resets all performance latches

sichtbeton, 2024-05-20:
### Please describe the problem:

Suggestion to make performance mode more intuitive when performing in grid mode:

steps to get there:
- enter perf view via pink button in grid mode
-  latch several columns
- exit perf via pink button again

now after this the following could happen:

- reset all column latches 
- go back to grid mode

### What is the expected behavior?

reset latches.

### Is there a relevant Pull request?

no

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

Exmpl-xe, 2024-08-06:
I think it'll be cool if optional  like holding the pink pad to clear the latches, tho this dose lose its immediacy

m-m-adams, 2024-08-11:
Maybe we could do it on release so you could time it with the long press?

closed 2024-12-28 by m-m-adams