SynthstromAudible/DelugeFirmware #1989: Filters can click when parameters are rapidly modulated.

sapphire-arches, 2024-05-21:
Large changes in filter cutoff or resonance cause most of the filter models to click, we should probably use crossfade logic like what was added added for #932 to avoid this. It would require rendering the filter twice for a window but maybe that perf hit is worth avoiding the clicks?

ok-reza, 2024-05-25:
I think this particular issue is expected based on the extreme modulation, I'm specifically noticing anything related to possible clicks is due to master level more than I am with filter/resonance.  I can replicate a similar experience you've detailed with other analog synths, (they just don't have access to increase osc1 level if it's already all the way up, unlike the Deluge). Rendering double the voices seems like it would add up, and this feels more like user-side problem/inexperience than a particular audio issue..

sapphire-arches, 2024-05-25:
yeah some of it is user error, but especially with automation view it's really easy to create automation that clicks

sapphire-arches, 2024-05-25:
maybe there's nothing we can do, but I think with some clever math we can bandlimit automation

m-m-adams, 2024-05-28:
Maybe we can put a temporary lpf on automation?

New value = (current value + automated value)/2 for like 2-3 render windows after a value change would probably do it

closed 2024-12-28 by m-m-adams