SynthstromAudible/DelugeFirmware #1993: Beta 1.1 Crash (buzz sound) when activating Fill (sync-scaling) on the second "fill" note of a MIDI kit row

soymonitus, 2024-05-22:
### Please describe the problem:

When you have a kit with midi rows (that send midi notes), then add some fill notes by holding SyncScaling and adding some notes to the grid, when you play the sequence and activate the fill mode, the first fill note plays fine, but the second note played will crash the deluge with a high volume buzz sound

### What is the expected behavior?

No crash

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

other

### What is the firmware name:

Beta

soymonitus, 2024-05-22:
After some more digging I found another freeze, which is if you hit Record button, it also hangs.

But this is all related to Midi Loopback Enabled. If I disabled Midi Loopback none of this happens. So steps to reproduce:
- First, enable midi loopback
- Create a new kit and add a midi row, and a sound row
- learn the sound row to the midi row
- Now press play and do either of these:
- Add fill notes to the midi row
- Engage Record

With any of these two actions the Deluge hangs

soymonitus, 2024-05-22:
@sapphire-arches

m-m-adams, 2024-05-22:
Seems related to https://github.com/SynthstromAudible/DelugeFirmware/issues/1817

The playhead is continually being reset to the same position

m-m-adams, 2024-09-22:
Fixed by removing loopback

closed 2024-09-22 by m-m-adams