SynthstromAudible/DelugeFirmware #1996: Recording Into New Kit Track Gives Card Error

ridemysailboat, 2024-05-22:
### Please describe the problem:

Not able to record into a kit row through the line in or mic in. It seems to work better with the internal mic, but recording with the line or mic in gives me the card error every time. When recording with the internal Deluge mic, it seems to work sometimes, ONLY if the audio level is clipping. If it doesn't clip, then it usually gives a card error. 

### What is the expected behavior?

Record into a new kit row and playback the sound without a card error.

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

deluge-v1_2_0-nightly+2024_05_21-a3c30ca

### If possible provide the steps to reproduce the issue and upload additional media:

1. Turn on
2. Create a new kit
3. Back out of sample select
4. Record into kit row
5. Card Error

m-m-adams, 2024-05-22:
Can you try the build from this PR? https://github.com/SynthstromAudible/DelugeFirmware/pull/1932

m-m-adams, 2024-05-22:
It's another case of a function infinitely looping to steal control, similar to song loading prior to 1932

https://github.com/SynthstromAudible/DelugeFirmware/blob/7c49bf1c766589294512eef996e480ee4b2a2814/src/deluge/gui/ui/audio_recorder.cpp#L186

This likely needs to be broken up in the same way. Notably the recording actually works so this is just a visual bug, it's correctly recorded into the RECORD folder under samples

m-m-adams, 2024-05-22:
https://github.com/SynthstromAudible/DelugeFirmware/blob/0163b22c5dd8bbfc64c81994afe599c4128ee320/src/deluge/model/sample/sample_recorder.cpp#L1489

@stellar-aria the problem is actually here, for some reason the lseek always returns an error

closed 2024-05-23 by sapphire-arches