SynthstromAudible/DelugeFirmware #2002: Modulation parameter menu should wrap around, or open at the top

nikodemus, 2024-05-23:
### Please describe the problem:

Previously modulation menu always opened so that ENVELOPE1 was in focus. Now it opens so that the first item in the menu that has a modulation value is highlighted instead.

This is convenient, but makes it harder to scroll through available options since eg. for cutoff MPE Y is assigned by default -- so one cannot scroll in one direction while thinking about options: instead you need to scroll first in one direction, then another.

Wrapping from top to bottom or vice versa would solve this.

### What is the expected behavior?

To be able to scroll through all modulation options without changing direction.

### Is there a relevant Pull request?

no

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

### What is the firmware name:

81eda8105de02842ce2c77df9d272a51cb15b63e

### If possible provide the steps to reproduce the issue and upload additional media:

Press `SHIFT + LPF`, press `SELECT`.

Observe MPE Y is in focus. Try to scroll through all modulation options.

nikodemus, 2024-06-03:
Thinking about this, for consistancy _all_ menu-like interactions should wrap around, no? Currently songs, do, but not much else.

sapphire-arches, 2024-06-03:
They do on 7seg, weirdly enough. There's a ton of code to implement both behaviors -- there have been discussions about changing this but nobody has committed to reworking it.

seangoodvibes, 2024-09-06:
Looking at this, the reason it doesn't open at the top is because it opens with the an existing patch cable source selected for that parameter.

Probably wrapping around would be the better solution than changing this behaviour?

[image]