SynthstromAudible/DelugeFirmware #2027: Can't exit Sound or Settings Menus by pressing anywhere on the mute/unmute/launch column when in either Keyboard View.

ok-reza, 2024-05-25:
### Please describe the problem:

Normally, the mute/unmute column is always available to press to exit any menu open on the Deluge. This is incredibly useful since the topmost mute/launch pad is right below the SHIFT button and is easily accesible.

With the new Keyboard View sidebar controls, this now doesn't allow users to access this shortcut when in a menu in Synth Keyboard View. Additionally, despite no sidebar controls being in Drum Keyboard View, the unused launch column also has no effect on exiting menus.

### What is the expected behavior?

Pressing a launch column should be consistent with normal Deluge behavior and exit the current menu before functioning as it's sidebar duties.

I have a feeling that this may slightly disrupt the use of the sidebar controls in Synth Keyboard View, as you may be editing a sound parameter via the Sound Menu and want to utilize auditioning notes along with the sidebar features. If the expected behavior was implemented, then while doing so you would then exit the open sound param menu for your first sidebar press, then resume as normal. However, the best way to find out if this feels natural and isn't bothersome is to implement it and see. 

Another option would be to include an 'EMPTY' sidebar control option where the given column has no use aside from exiting a menu (if it's the launch column?). However, this means that a user won't be able to utilize a consistent Deluge feature unless they set their sidebar column each time they boot up, which is undesirable.

### What hardware did you reproduce it with?

OLED, 7SEG

### What firmware did you reproduce it with?

other

### What is the firmware name:

1.1 beta

### If possible provide the steps to reproduce the issue and upload additional media:

Synth Keyboard View:
1) Enter Synth Keyboard View
2) Enter a param shortcut (SHIFT+LPF)
3) Hit the first sidebar column to exit sound parameter menu.
4) No exit is made, velocity may adjust to 127.

Drum Keyboard View:
1) Enter Drum Keyboard View
2) Enter a param shortcut (SHIFT+LPF)
3) Hit the first sidebar column on the left to exit sound parameter menu.
4) No exit is made.

sichtbeton, 2024-05-27:
Yes, 100% agree.

sichtbeton, 2024-05-27:
The "mute" button column function could be re-introduced.

ok-reza, 2024-05-29:
I understand how that would be undesirable for that use case.

But yes, unlearning the muscle memory and using BACK instead of the mute column closest to SHIFT (where my hand naturally rests as I edit) is not ideal. Especially since using that column to exit will still be available everywhere else on the Deluge. It's displeasing enough that if an empty sidebar option is introduced, I'd likely set it so the left sidebar is default empty and I'd only have access to one sidebar control at a time. But ideally, I could have two sidebars available to me at given time and have the expected behavior I detailed of exiting when a menu is open.

Outside of sound param editing menus, I think it should still be able to exit first when in Deluge Settings Menus as that shouldn't disrupt the use case you describe and is universal on the Deluge.

ok-reza, 2024-06-22:
1 month update:

This isn't something muscle memory can adjust to since this ability is possible everywhere else on the Deluge.

Additionally, depending on how deep you are in sound parameters or menus, it would take multiple BACK presses to actually exit compared to what normally is always just one press with where the Sidebar columns are.

This needs to have an option to enable exiting menu's before sidebar controls are registered.

closed 2024-06-26 by m-m-adams