SynthstromAudible/DelugeFirmware #2036: Sample path as `setting name` in CommunityFeatures.XML

scy, 2024-05-26:
### Please describe the problem:

I've just noticed that I have these lines in my `CommunityFeatures.XML`:

```xml
	<setting name="emulatedDisplay" value="0"></setting>
	<setting name="SAMPLES/Artists/Danny Taurus/JusBeats/Spiky-NoKS-120.wav" value="1"></setting>
	<setting name="SAMPLES/Artists/Danny Taurus/JusBeats/Spiky-NoKS-120.wav" value="1"></setting>
	<setting name="SAMPLES/Artists/Danny Taurus/JusBeats/Spiky-NoKS-120.wav" value="1"></setting>
	<setting name="SAMPLES/Artists/Danny Taurus/JusBeats/Spiky-NoKS-120.wav" value="1"></setting>
	<setting name="SAMPLES/Artists/Danny Taurus/JusBeats/Spiky-NoKS-120.wav" value="1"></setting>
</runtimeFeatureSettings>
```

Is that, uh, expected behavior to have the path to a sample as a **setting name** in the XML? :thinking:

### What is the expected behavior?

No random file paths used as setting names, especially not multiple times the same one ;)

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

other

### If possible provide the steps to reproduce the issue and upload additional media:

Sorry, I can't exactly tell you which firmware version created this issue, as I usually update weekly to the latest beta.

The last time I've used that sample was on May 5. If it's super useful, I can try to reconstruct what firmware version I've most likely used at that time, but please only make me do that if it's really relevant. Maybe you're instead already like "yeah, that was a bug and we've reverted it in the meantime".

The `CommunityFeatures.XML` also mentions the song I was working on at that time:

```xml
<runtimeFeatureSettings
	firmwareVersion="c1.1.0"
	earliestCompatibleFirmware="4.1.3"
	startupSong="SONGS/SCY/WeeklyBeats/WB2418 Ugh.XML">
```

Interestingly enough, I've since worked on other songs, but they don't seem to be considered for `startupSong`. Maybe that gives you an additional hint on what caused the problem.

sapphire-arches, 2024-05-26:
have you only been running beta channel builds or do you use nightly at all?

scy, 2024-05-26:
Might have also used some nightly releases, yes.

sapphire-arches, 2024-05-26:
OK we completely rewrote the file IO code in nightly so I'm going to guess that's the problem. I'll see if I can replicate.

m-m-adams, 2024-07-21:
I think this is fixed? I think it was the buffer reuse thing in the storage manager

closed 2024-07-21 by m-m-adams