SynthstromAudible/DelugeFirmware #2055: Song loading and audio file loading can interfere with eachother causing crashes

sapphire-arches, 2024-05-30:
### Please describe the problem:

With the following patch:

```diff --git a/src/deluge/gui/ui/load/load_song_ui.cpp b/src/deluge/gui/ui/load/load_song_ui.cpp
index 87ad067e50...44453cf331 100644
--- a/src/deluge/gui/ui/load/load_song_ui.cpp
+++ b/src/deluge/gui/ui/load/load_song_ui.cpp
@@ -246,6 +246,8 @@
        Song* toDelete = nullptr;
        // static LoadStatus status = LoadStatus::START;

+       D_PRINTLN("Load state %d", status);
+
        switch (status) {
        case LoadStatus::START: {
                FileItem* currentFileItem = getCurrentFileItem();
@@ -378,14 +380,14 @@
                        preLoadedSong->loadAllSamples(true);
                }
                status = LoadStatus::BEGIN_LOADING_SAMPLES;
-               addOnceTask([]() { loadSongUI.performLoadFixedSM(); }, 100, 0.05, nullptr);
+               addOnceTask([]() { loadSongUI.performLoadFixedSM(); }, 100, 1.000, nullptr);
                break;
        }
        case LoadStatus::BEGIN_LOADING_SAMPLES: {
                // Ensure all AudioFile Clusters needed for new song are loaded
                // Prevent any unforeseen loop. Not sure if that actually could happen
                if (audioFileManager.loadingQueueHasAnyLowestPriorityElements() && count < 50) {
-                       addOnceTask([]() { loadSongUI.performLoadFixedSM(); }, 100, 0.05, nullptr);
+                       addOnceTask([]() { loadSongUI.performLoadFixedSM(); }, 100, 1.05, nullptr);
                        return;
                }

@@ -434,7 +436,7 @@
                        AudioEngine::logAction("h");
                        // If any more waiting required before the song swap actually happens, do that
                        if (currentUIMode != UI_MODE_LOADING_SONG_NEW_SONG_PLAYING) {
-                               addOnceTask([]() { loadSongUI.performLoadFixedSM(); }, 100, 0.05, nullptr);
+                               addOnceTask([]() { loadSongUI.performLoadFixedSM(); }, 100, 1.05, nullptr);
                                return;
                        }
                }
@@ -445,7 +447,7 @@
        }
        case LoadStatus::WAIT_FOR_SAMPLES: {
                if (currentUIMode != UI_MODE_LOADING_SONG_NEW_SONG_PLAYING) {
-                       addOnceTask([]() { loadSongUI.performLoadFixedSM(); }, 100, 0.05, nullptr);
+                       addOnceTask([]() { loadSongUI.performLoadFixedSM(); }, 100, 1.05, nullptr);
                        return;
                }
 swapDone:
```

And [this song file](https://discord.com/channels/608916579421257728/1138924035657449482/1245639164070858875) we reliably crash from cold boot (at least when based on commit 2a678fd33be5e3ecc0e2dbe5609366e053c9718f from PR #2054).

It seems to be some data race where a buffer overrun causes us to stomp some vtable pointers, as the fault address is an attempt to call `reader.exitTag` (assuming the system is even in a state where we can trust the black box).

### What is the expected behavior?

no crash =)

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

m-m-adams, 2024-06-02:
It's possibly from changing UI modes and not from audio file loading specifically

m-m-adams, 2024-06-07:
This should be fixed - I'd guess it was the timing of ui rendering fixed by #2081

closed 2024-06-07 by m-m-adams