SynthstromAudible/DelugeFirmware #2075: Saving a Kit Row sample as Synth saves the new synth to the Project folder (not Synth folder, unable to load "synth" to Kit row or Synth)

thejesiah, 2024-06-02:
### Please describe the problem:

Trying to utilize the ability to save/load samples-as-synths to Kit rows. I create a new Kit. Load a sample. Audition+Save the Kit Row as a Synth with new name.. it plays and displays the new name, seemingly as normal.  
But, on a new Kit Row, Audition+Load, the new Synth is nowhere to be found. 

On further investigation, the new synth appears in the Load Song menu, but displays FILE CORRUPTED

Note that I was just now noticing this on a Nightly installed on May 8th. Updated to tonight's nightly and problem is the same.

Thank you for your help on this! 

### What is the expected behavior?

For the newly saved synth file to be saved to the synth folder and to be the immediate, first option when loading a new Kit Row Synth

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

deluge-v1_2_0-nightly+2024_06_02-7339263

### If possible provide the steps to reproduce the issue and upload additional media:

Turn on, New Song
Shift + Kit to create new Kit and browse to a preferred sample, Select 
Audion+Save to save synth & rename (I've been renaming mine "X name"), & Save
choose a blank Audion+Load ... try to find the new synth. nope. 
Back out to top, hit Load (Song), and scroll to name of new synth.. will display File Corrupted

m-m-adams, 2024-06-02:
Are you sure that the firmware update worked? This should have been fixed by #1964 a couple weeks ago

thejesiah, 2024-06-02:
Derp, just tried the firmware update again and tested and worked... thank you!

closed 2024-06-02 by thejesiah