SynthstromAudible/DelugeFirmware #2077: startup song loading is broken on community branch

nikodemus, 2024-06-02:
Having a startup song set on community branch makes deluge crash on startup.

m-m-adams, 2024-06-02:
The main issue seems to be not setting the folder storage manager path. The song xml is found by searching the current folder for the song name. This is particularly bad if the current folder isn't the song folder

Another issue is here https://github.com/SynthstromAudible/DelugeFirmware/blob/2ee2b695577964558727a21a39e2c0a248a61170/src/deluge/deluge.cpp#L499,  which breaks audio rendering while the song is being loaded

The third issue is that if the SD card wasn't ready then it would fail to read/write the canary file and get stuck in a loop. That's fixed by making the startup song conditional on the SD card being ready

They all combine to make the boot loop

nikodemus, 2024-06-03:
Confirmed as working now. 🍾

closed 2024-06-03 by nikodemus