SynthstromAudible/DelugeFirmware #2104: Deleting Phantom Kit Presets Fail

acook, 2024-06-07:
### Please describe the problem:

While cleaning up some drum kits I ran into this unexpected error which I haven't seen mention anywhere and no search results appear. When trying to delete kit presets, under certain conditions, I receive the message "Error Deleting File".

The first time this happened with 3 kits, 2 were empty, and 1 had only a single sample. I couldn't remove those kits at all. But after restarting the Deluge they were all gone.

It happened again an hour of kit making later with kits that seemed to spontaneously appear every time I wanted to add a sample. The fact that it has happened multiple times, with a reset in between, tells me that this is a problem I'm going to keep running into.

### What is the expected behavior?

It shouldn't create phantom kits and even if it did, it should allow them to be removed from memory.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

other

### What is the firmware name:

4.1.4 (official)

### If possible provide the steps to reproduce the issue and upload additional media:

1. Create several kits until phantom kits start being created
2. Delete phantom kit (load+kit, then scroll to kit and press shift+delete then delete)
3. Receive error, kit remains

> Error Deleting File

---

I'm building kits with generally 16 slots filled with one-shot samples. I've made around 30 kits today, not counting ones that got lost when I reset, so I can guess that the issue appears when around 16 new kits have been created in one sitting.

So while it is inconsistent, it seems to happen after making several kits. So it may be related to sample memory or some kind of internal pointer when adding new kits?

I think the error happens because those phantom kits don't really exist, they haven't been saved to the drive, so their files can't be deleted.

acook, 2024-06-07:
[image]

m-m-adams, 2024-06-09:
We don't support the official firmware here - does this occur on community as well?

acook, 2024-06-12:
Ah, I must have misread the info on the site, I thought this was the place for bug reports for the official firmware.

I'm not sure if it happens on the community build too, the code were that error is displayed has been changed, but I don't think it's the deletion code that's the problem. 

I've tested it further with a different set of samples and it didn't happen, the second set of samples was a lot smaller overall, so that provides more evidence that the problem is related to sample memory somehow. 

I have switched to the community firmware ([Release 1.1 RC 2](https://github.com/SynthstromAudible/DelugeFirmware/releases/tag/release_1_1_0_rc2)) and will continue to test and report my findings.

m-m-adams, 2024-06-12:
The file manager was changed to fix a bug with incorrectly loading presets after changing scroll direction a while back, this might be a symptom of the same underlying bug and also be fixed by coincidence

acook, 2024-06-18:
No updates on this, unfortunately. I will pick it up again next week

PR #2401 (2024-08-08): Bugfix/if instrument not on card delete it from hibernation on change
Don't want to cherry pick this just in case I missed something but it's almost certainly good

Fix #2104
closed 2024-08-08 by seangoodvibes