SynthstromAudible/DelugeFirmware #2110: Sticky shift should exclude audition / mute rows to facilitate ongoing changes while auditing.

mnantel, 2024-06-10:
### Please describe the problem:

This is an enhancement suggestion, well described by the title.
Sticky shift is very useful when working on numerous shift heavy tasks.
However when it is active, the audition buttons light up when pressed but no audio is produced.

### What is the expected behavior?

It should be possible to audition while sticky shift is active.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

other

### What is the firmware name:

1.1.0-BETA-C2EC246

m-m-adams, 2024-06-10:
This has been discussed a number of times, but ultimately we've decided on consistency and making sticky shift behave exactly like holding shift. Keep in mind that some people do want it to mute the audition pads as well - this is great for editing sounds while a sequence is playing back. There's some inconveniences with this approach but we think it's better than having confusion about what sticky shift applies to

m-m-adams, 2024-06-10:
Leaving this open for a bit for discussion

mnantel, 2024-06-10:
One option might be to subtree the community feature menu, such that this behaviour can be toggled under the "sticky shift" menu. I understand the need to avoid complexity.