SynthstromAudible/DelugeFirmware #2127: saving synth from kit saves towering location

lmcmulle, 2024-06-14:
### Please describe the problem:

synth is saved in the “songs” path not the “synths” and is thus unloadsble

when saving a synth from a kit by pressing shift-aud-save 

### What is the expected behavior?

should save in synths path 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

from about 2 weeks ago

sapphire-arches, 2024-06-14:
Please retest with a newer nightly, we fixed a bug related to this recently.

m-m-adams, 2024-06-18:
Any update? This seems fixed when I check it

lmcmulle, 2024-06-20:
tested the jun19 nightly build this resolved.

closed 2024-06-20 by m-m-adams