SynthstromAudible/DelugeFirmware #2132: CV/Gate clock output fails after long uptime

acook, 2024-06-15:
### Please describe the problem:

If the Deluge is on for long periods of time the gate output clock will send a single pulse and then freeze.

### What is the expected behavior?

Regardless of uptime, the gate clock will continue functioning.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.0/Amadeus

### What is the firmware name:

1.1.0-BETA-E45865D

### If possible provide the steps to reproduce the issue and upload additional media:

1. Turn on
2. Configure CV/Gate output for clock mode
3. Observe functional clock output
4. _Wait_ (easy mode: leave it on overnight)
5. Press play
6. **Observe failure to send proper clock**
7. Restart Deluge
8. Observe functional clock output

---

This problem has also existed on the official firmware for as long as I can remember.

Probably the same underlying issue or at least related to: https://github.com/SynthstromAudible/DelugeFirmware/issues/2123

m-m-adams, 2024-06-18:
Confirmed, sadly at one debug attempt per 24 hours or so this will probably take a while to fix...

m-m-adams, 2024-06-18:
I think this might be the same as 2144 - they both end up in a state where clockOutputPending isn't being set properly. 2144 is (poorly) fixed by having CV output handled exclusively by the ISR but this state isn't supposed to be possible anyway

[image]

As shown here, they don't match so a clock should be pending - it's unclear how this happens

m-m-adams, 2024-06-18:
In fact all combinations seem to be possible 

[image]

m-m-adams, 2024-06-23:
Duplicate of #2144, only difference is the clock starts to fall behind faster in nightly. Once the clock is behind it outputs 2 ticks at once to catch up which results in the gate not changing at all.

closed 2024-06-23 by m-m-adams

m-m-adams, 2024-07-07:
Confirmed this is now fixed