SynthstromAudible/DelugeFirmware #2134: Unlinking cloned clips fails

rivo, 2024-06-15:
### Please describe the problem:

A cloned clip (synth or kit) could be unlinked as described in the Guidebook of which I've pasted a screenshot below. In 1.1 RC2, this does not work. Here's how to reproduce this.

1. Create a new clip, for example a synth in clip view, with SHIFT+SYNTH, which will then be called "SYNTH 1". (The problem is exactly the same for kits but I'm using synths as an example here.)
2. Clone the clip in song view, by holding a row pad of that clip and then pushing a pad in an empty row.
3. These two synths are now linked, i.e. only one will play at a time.
4. Open the cloned clip by clicking on a pad.
5. Hit LOAD+SYNTH. Menu will (amongst other synths on the SD card) show the new synth, "SYNTH 1".
6. Make sure to select that synth by turning the SELECT knob to "SYNTH 1".
7. Press & hold LOAD.
8. The display will now say "FILE NOT FOUND". The synths will not unlink. Everything will be the way it was before.

Here's what _should_ happen (this is from the Guidebook):

[image]

(By the way, item 4, _will show 'USEd'_ also does not happen. But I suppose this is less of a problem.)

### What is the expected behavior?

The display should ask me "Clone?" or something similar, after which I should be able to press LOAD again, to obtain an (unlinked) copy of my clip.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

other

### What is the firmware name:

1.1 RC2

m-m-adams, 2024-06-16:
This doesn't work unless you've saved the synth you're trying to unlink. The cloning mechanism is a reload from the SD card and your file is not there yet (e.g. FILE NOT FOUND)

m-m-adams, 2024-06-16:
arguably needs a better error message

rivo, 2024-06-16:
This is odd because my (unsaved) synth appears in the list. So it's not just the error message that is confusing.

The Guidebook section which I pasted above also doesn't say that the clip needs to have been saved first. Does this mean the documentation here is incomplete?

Finally, what is the correct way to _clone and unlink_ a clip that has not been saved?

m-m-adams, 2024-06-16:
Yeah all presets in the song appear in the list regardless of whether they've been saved. They can be linked to a clip but not cloned. This is a longstanding issue that will probably be changed eventually to note whether the preset exists on the card or just in memory

It's not possible to clone without saving, just save it first. The documentation is incorrect here