SynthstromAudible/DelugeFirmware #2139: Long samples cutting out

Lugaseurop, 2024-06-16:
### Please describe the problem:

A very important feature of the Deluge for me is the ability to playback long samples. I am regularly but inconsistently finding that long samples cut out - this is even when a single sample is triggered from a kit with no other activity happening on the deluge. When the sample is played, it plays through until around 50% of the full sample and then just cuts out. 

### What is the expected behavior?

In the official firmware, the sample reliably plays the whole way through.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community, other

### What is the firmware name:

Beethoven Beta release 1.1

### If possible provide the steps to reproduce the issue and upload additional media:

1. turned on
2. loaded track
3. triggered sample from kit
4. sample plays for around 50% of the track (different each time) then cuts out.

seangoodvibes, 2024-06-16:
@Lugaseurop can you share your song with the samples? e.g. save with collect all?

Thx!

m-m-adams, 2024-06-16:
There's a mention from the official firmware of a known bug around a single long sample playing back - https://github.com/SynthstromAudible/DelugeFirmware/blob/b5c51ea7c5c0977efbe6e5618e60c089b8971952/src/deluge/memory/cache_manager.cpp#L76

could that be the case here? There might be a regression in how its handled

Lugaseurop, 2024-06-27:
Hey 

Thanks for getting back to me and for your patience.
Here are a couple of songs I've had trouble with.
Thank you!
Lewis
https://drive.google.com/drive/folders/13otL-2SW14CKFt3zsyLcN9dgYheqN6DZ?usp=sharing

m-m-adams, 2024-07-04:
Sorry these are missing the song xml files, could you share those too so I can load them? Thanks!

m-m-adams, 2024-07-27:
Will reopen if a song that reproduces the issue is shared

closed 2024-07-27 by m-m-adams