SynthstromAudible/DelugeFirmware #2144: Severe Trigger Clock Dropouts

acook, 2024-06-17:
### Please describe the problem:

Alternate title: Trigger Clock Output Fails to Meet Realtime Deadline

When using the analog clock at the gate output, it will randomly stop transmitting for long periods.

On 8aa506f this happens after a few seconds. According to @m-m-adams it happens after a few minutes on **Release Candidate 3** 564cba31cd14d9fbfc16a28114ab6d4dba9441e6.

### What is the expected behavior?

The trigger clock putters along more-or-less constantly, within a margin of error that most humans (and downstream sequencers) consider indistinguishable from lock-step. 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

8aa506f

### If possible provide the steps to reproduce the issue and upload additional media:

1. Turn on
2. Configure output 4 for clock
3. Set desired PPQN (happens at any but more obviously with larger PPQN)
4. Hook up something to watch the clock output
5. Press play
6. Wait a few seconds (typically less than 30) and the clock output will stop and then restart, this will continue to happen over and over again at apparently random intervals

## Additional Info

Tested with multiple downstream devices, but it is so severe that it is visible with the naked eye.

Watching a blinking LED connected to the output is sufficient to detect the dropouts at higher PPQNs, it is that major! The LED (or any downstream sequencer/clock divider) will receive no signal at all for up to nearly a second(?!) before the Deluge resumes transmitting, leaving the downstream devices hopelessly out of sync.

The improvements between the latest nightlies and the current community branch is likely due to improvements in the yield/scheduler code that have gone in, but it remains. I'm not sure what commit introduced the issue.

**NOTE: This is *unrelated* to the other analog clock output bug that I reported.** 

- This happens *immediately* after being powered on and presents as randomly the clock not outputting and then resuming. 
- This bug does not occur with the official firmware.

m-m-adams, 2024-06-23:
When the clock falls behind it attempts to catch up by sending two clocks at once, resulting in inconsistent midi clock and no gate output.

The clock falls behind when two ticks need to be output in the same render window. As the nightly build is targeting larger consistent window sizes this now occurs more frequently, however it is present in official and the RC as well (#2132 and unstable midi clock reports on discord).

The fix will likely be to proactively schedule ticks further in advance

m-m-adams, 2024-07-05:
I think I've got it - if you could test the build from the linked PR that would be great, but it's been stable for over an hour now

PR #2145 (2024-07-06): Don't send multiple clocks at once
Fix #2144 

Bug occurs because clocks are only scheduled when within numSamples worth of time at the end of the audio routine, but this causes a bug because the next render call is not guaranteed to be before numSamples have passed (there's about 64 samples of extra buffer on average). This would not occur during the main loop in official but was possible in routineForSD, making the bug take much longer to reproduce on official and 1.1. As the sequencer is only updated for the next numSamples it cannot be scheduled further in advance without sequencer changes [... cut]
closed 2024-07-06 by sapphire-arches