SynthstromAudible/DelugeFirmware #2151: Deluge “Freezing” when duplicating a clip

lmcmulle, 2024-06-21:
### Please describe the problem:

scenario: playing an arrangement under moderate to heavy cpu load duplicating a clip will cause the machine to freeze requiring a power cycle 

intermittent problem - happened twice with the jun 19 nightly build- only while playing - but not every time.  i will run some tests to see if i can narrow the fault conditions 

L

### What is the expected behavior?

clip will duplicate while playing and not freeze the machine 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

jun 19th nightly 

### If possible provide the steps to reproduce the issue and upload additional media:

playing an arrangement with several tracks placing the machine under moderate to heavy load (no apparent voice stealing ) 
duplicating a clip of a synth track which is playing caused the machine to freeze

lmcmulle, 2024-06-21:
issue persists with Jun-21 nightly -
- issue is now replicated across multiple tracks
- high cpu light is NOT flashing 
- only occurs when sequencer is playing (it seems) 
- if the sequencer is playing - and the track being doubled is not playing  - the issue still occurs 
- there does not seem to be a pattern to when it happens sometimes it’s in the first, but as many as 4 doubles before it happens 
- -track length of the track being doubled  does not affect behavior - 
- tested to see if resulting size (eg 4,8,16 bars) was a factor it was not -

lmcmulle, 2024-06-22:
additional testing:
- doubling one clip - backing out to arranger - doubling a variation of the same clip - freeze occurs 
- double one clip - switching instruments - double that clip - freeze occurs 

###net it out: doubling the clip causes the freeze - somewhere between the 2nd and 4th double - regardless of instrument /track changes or  other state of the machine eg playing or not-

m-m-adams, 2024-06-25:
I have had this happen so it's definitely a bug. I'm not having luck deliberately reproducing while attached to a debugger so far though.

m-m-adams, 2024-06-26:
Most reliable repro - make a couple tracks using large samples (like BoD), add notes, press play, then alternate doubling tracks until a freeze happens

Freeze happens in memcpy in resizeableArray::generateRepeats which appears to clobber everything 

Non optimized builds do not freeze

I suspect the compiler upgrade picked up on some UB because this code appears to all be original

m-m-adams, 2024-06-26:
This should now pop up either D001 or D002 - if you get the freeze without the error message please post again, I don't see a root cause yet so I want to be sure this is the culprit

lmcmulle, 2024-06-29:
i can induce the error -it just freezes -  but no code shows - i’m not connected to a debugging environment - 

happy to do that if someone points me to how to connect it up to a debugger

m-m-adams, 2024-06-29:
You need a hardware debug probe for proper debugging but you can get printouts over midi using dbt (our build tooling). I think the command is dbt sysex logging but the full instructions are in contributing.md 

So just to confirm you're on a new nightly build (27th or later) and not getting the error message? It's definitely memory corruption, the error catches an issue between the popup holding <> and the clip doubled message overwriting a parameter that gets used in a later memcpy

It's possible there's another situation where it occurs too

matreve, 2024-06-29:
Still freezing with 6/29 nightly. The D0001 and D0002 pop up, but not always, often it goes straight to the „crazy LEDs pattern” screen right away. It’s never possible to restore normal operation, even when codes are displayed.

m-m-adams, 2024-06-29:
Oh perfect yeah - if it's getting the freeze screen without the code showing that's catching it. Showing the error is probably inconsistent since the corruption is in the display memory.

The memory is already corrupted at that point so it's not recoverable, it's just freezing before it hits the infinite loop. Hopefully one of the devs that's worked on display code can fix it

matreve, 2024-06-29:
Not sure it this is relevant but there’s also a faint high pitch sound when the freeze happens. It sounds a bit like very short cycle granular synthesis. Is anyone else also experiencing it? Infinite loop makes sense.

m-m-adams, 2024-06-29:
Yeah the output dac will just loop its 128 sample buffer when it stops getting new input

lmcmulle, 2024-07-01:
definitely still happening with the 6/30 nightly - even a single kit/track is causing it (though sometimes you need to double more than once )

m-m-adams, 2024-07-01:
Do you get the lights? Or just nothing at all

lmcmulle, 2024-07-04:
sorry for the delay … i really appreciate your efforts - i get “D002 press select… “ and a a light pattern on the grid now

sapphire-arches, 2024-07-04:
this shouldn't crash still.

matreve, 2024-07-04:
Still crashes exactly the same way.  Why was this issue closed? Really just trying to understand the logic behind closing this issue.

m-m-adams, 2024-07-04:
> Still crashes exactly the same way. Why was this issue closed? Really just trying to understand the logic behind closing this issue.

The author closed it, not any of the devs. All we've done so far is catch it on an error screen

matreve, 2024-07-04:
> > Still crashes exactly the same way. Why was this issue closed? Really just trying to understand the logic behind closing this issue.
> 
> The author closed it, not any of the devs. All we've done so far is catch it on an error screen

I understand it. My point is, this thread is about a common issue that many users experience, not just the author. Author needs to realize it.

m-m-adams, 2024-07-04:
People are allowed to make mistakes. Please respect our [code of conduct](https://github.com/SynthstromAudible/DelugeFirmware/blob/community/docs/CODE_OF_CONDUCT.md)

matreve, 2024-07-04:
> People are allowed to make mistakes. Please respect our [code of conduct](https://github.com/SynthstromAudible/DelugeFirmware/blob/community/docs/CODE_OF_CONDUCT.md)

Oh, I did not mean to make it sound so serious. Apologies if I did. If anything, trying to learn how to navigate this github environment. I don’t contribute much other than whining. And if the Deluge crashes using this free code, it’s really not that important, really.

lmcmulle, 2024-07-08:
I did not intend to close the issue. I'm sorry I was working on my phone and must have hit the wrong button

lmcmulle, 2024-07-18:
This resolved the issue for me... I've tested about half a dozen different scenarios... the deluge is no longer freezing when I double clips... Closing.

closed 2024-07-18 by lmcmulle