SynthstromAudible/DelugeFirmware #2243: Automate Resampling of Clips for "Simulated Stem Export"

seangoodvibes, 2024-07-05:
### Please describe the problem:

The manual way of exporting clip stems is to resample each clip individually.

### What is the expected behavior?

Add a special playback mode that automates the resampling of clips individually. This special playback mode would not let you do anything else. You would wait until it is done resampling all clips (e.g. display could show # of clips processed / remaining). You could however cancel out of the mode to interrupt the process.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

sichtbeton, 2024-07-06:
Omg. This. Is. So. Good.

sichtbeton, 2024-07-06:
Complete tracks (for each instrument) would also be sufficient for most I think. Individual clips is nice but it would cut off reverb tails or delays in arrangements.

seangoodvibes, 2024-07-06:
> Complete tracks (for each instrument) would also be sufficient for most I think. Individual clips is nice but it would cut off reverb tails or delays in arrangements.

Starting with clips and then we can go from there.

m-m-adams, 2024-07-06:
Converting to discussion - using issues for ideas makes it easy to miss bug reports