SynthstromAudible/DelugeFirmware #2262: fast initiating resampling in arranger view misses the first playback of a sample (and doesn't resample it)

sichtbeton, 2024-07-07:
### Please describe the problem:

When trying to resample this song and pushing the combo "record" + "play" fast, it doesnt play the main sample for the first time. it does catch up and play it the second bar around. contrary when doing the combo slower, it works as expected.

here is a video for it:

https://github.com/SynthstromAudible/DelugeFirmware/assets/62035517/51992418-9e76-445e-83ef-8b8e02fac01b

the song in question:

https://drive.google.com/file/d/14xGaZgkDKFA7WlozfBc3il-QEXsaMz-p/view?usp=sharing

### What is the expected behavior?

play sample from beginning and resample correctly

### Is there a relevant Pull request?

no

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

### What is the firmware name:

1.2.9-DEV-5C8AF7B

### If possible provide the steps to reproduce the issue and upload additional media:

- load song
- go to arranger
- press record and play fast

m-m-adams, 2024-07-08:
It's being culled - appears to be a pre existing bug from official, there's a word of Rohan comment about attempts to fix it in AudioRecorder::beginOutputRecording()

sichtbeton, 2024-07-08:
Oh, right so it’s a rather deep issue I presume?

m-m-adams, 2024-07-08:
Just needs investigation. The function rohan commented in isn't the one causing the issue so someone needs to figure out which one actually is