SynthstromAudible/DelugeFirmware #2268: Arp gate is shorter in c1.1 when ENV1 Release is set to 0 or near 0

ok-reza, 2024-07-08:
### Please describe the problem:

If env1 release is 0 or near 0, arp gate amount generates more choked sounding notes when compared to previous firmware.

This is most apparent when release is set to 0, for 'SONG13G', provided by the user, setting release to 3 or 4 unified the gate behavior between firmwares.

SONG13G files: https://www.dropbox.com/scl/fo/krncct3qgqwsgh3o52jay/AHZR9RCWnariXl1gSSgHaWU?rlkey=pzjvbosak38toi6zdu5ij961c&e=1&dl=0

### What is the expected behavior?

Unify arp gate between firmware versions at low release values.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.1/Beethoven

### What is the firmware name:

c1.1

### If possible provide the steps to reproduce the issue and upload additional media:

1. Create new synth, new project, bpm 120, set ENV1 Release to 0.
2. Enter a drone note across 1 bar
3. Set synth Arp Mode to ON (any arp mode should do), set Arp Rate to 32nd notes
5. Playback
6. Set Arp Gate to lower values, like 20, 10, or 5.
7. Resample these arps, load previous firmware, and compare them to the resampled versions.

m-m-adams, 2024-07-21:
Which track in this song has the problem? There's a lot of them

ok-reza, 2024-07-21:
it's one of the arp tracks, i think in arranger view it's towards the bottom, 2nd to last? if i recall
it was named synt17. can confirm in a couple hours

m-m-adams, 2024-08-08:
I've poured through this tonight and I don't think it's a bug, probably a result of more accurate window sizes not smearing the gate and release

Maybe we should bump up the minimum gate so it always at least starts a note?

soymonitus, 2024-12-07:
[image]
Release times are artificially longer on 1.0 vs 1.2.
This is an synth with Env1 release = 0. It is doing a fade in 1.0 as if release was 1 or 2 instead of 0

seangoodvibes, 2024-12-07:
[image]

I think I'm good with concluding that we fixed a bug in 1.1 that was present in 1.0 and probably official also.

A sound should not be releasing if the envelope release is 0.

I don't think it's related to the arp.

stellar-aria, 2024-12-10:
Agreed, I don't think there should be any release time if the release time is set to 0. I think we can close this as not a bug.

closed 2024-12-10 by seangoodvibes