SynthstromAudible/DelugeFirmware #2274: USB MIDI Clock output is roughly doubled and inconsistent

konkrete, 2024-07-10:
### Please describe the problem:

Example:
Chosen tempo in Deluge is 100bpm, sent factual clock red by ShowMIDI is 195.8.
Strange one: Automatically detected tempo by synced Bitwig is 100bpm, however factual playback of clips in Bitwig is much faster and looks and sounds like almost 200bpm, that is confirmed by ShowMIDI.

### What is the expected behavior?

Expected working properly, sending proper clock signal.

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Release 1.1/Beethoven

### What is the firmware name:

c1.1.0

### If possible provide the steps to reproduce the issue and upload additional media:

Turn On, press Play.

sapphire-arches, 2024-07-10:
Sounds like showmidi is listening to both virtual cables. Either select only one port, or disable clock output on one (under settings>midi>devices)

m-m-adams, 2024-07-10:
Confirmed show MIDI doesn't differentiate between the cables. You can stop the extra clock on the deluge side by going to midi->devices->upstream2->clock and set to off, or you can make sure your daw is only receiving clock from one port

closed 2024-07-10 by m-m-adams

m-m-adams, 2024-07-11:
@konkrete if you're on mac remove and re add the deluge in MIDI studio