SynthstromAudible/DelugeFirmware #2275: I don't understand how to MIDI learn a keyboard in the new firmware

miaouxmiaoux, 2024-07-10:
### Please describe the problem:

Tested with the official firmware, so I know MIDI routing is correct. I've watched Ron C's video on it and in MIDI Follow -> Follow -> Channel A his screen has "DIN Ports" on the second row while my Deluge says "Device Unassigned". Holding learn and pressing a key on the keyboard I want to use, as he does, does nothing. The old workflow - holding learn and pressing the 'Audition / Section' row - does nothing. What am I doing wrong?

### What is the expected behavior?

I want to MIDI learn a keyboard

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.1/Beethoven

sapphire-arches, 2024-07-10:
You should still be able to learn midi devices to clips the same way. To learn a MIDI follow channel, hold learn/input while in that menu and press a key on the controller.

miaouxmiaoux, 2024-07-10:
Thanks, neither of these are working. It must be a problem with my setup, but it's odd because MIDI learn works with the official firmware but not Beethoven. So all the routing is correct and the Deluge is receiving MIDI from the USB port. But as soon as I load Beethoven the Deluge no longer receives MIDI over USB.

m-m-adams, 2024-07-10:
Are you on a mac or iPad? If so it's an Apple bug and you need to reset your MIDI settings to fix it

miaouxmiaoux, 2024-07-10:
Thanks! I'm on a Mac. The Deluge and all MIDI devices are connected via an iConnectivity mioXL router.

I'll give this a go, what exactly do you mean by 'reset MIDI settings'?

m-m-adams, 2024-07-11:
Oh if you're on a mioXL remove the deluge from USB host reservation, restart the mioXL, then re add deluge port 1

[image]

should look something like this afterwards. We added extra virtual ports but the mioXL won't see them yet