SynthstromAudible/DelugeFirmware #2290: Sticky shift won't stop stickin'

madeline-scyphers, 2024-07-12:
### Please describe the problem:

Turned sticky shift on, sticky shift would not turn off. Tried pressing shift quickly several times to get out, would not work. Then pressed some hard and slow ones thinking maybe it is my button itself and it just needs to TLC from time to time (which might be the case), and then pressed fast again, which did not get me out of it. I ended up being locked in sticky shift for about 5 minutes trying to get out, moving around my deluge menus to see if anything would break me out before something finally got me out.

https://github.com/user-attachments/assets/7e326555-ae35-4f87-819e-df1d2f5a59dd

### What is the expected behavior?

Quickly pressing shift will turn sticky shift off.

### Is there a relevant Pull request?

No

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

### What is the firmware name:

c1.1

### If possible provide the steps to reproduce the issue and upload additional media:

I can't reproduce it, but some possible relevant things are that I have set my `hold press time` to 100 ms, which may make it more likely. I also had done several undo and redo in a row before this, but in a new song that doesn't reproduce it.

m-m-adams, 2024-07-15:
10/10 title

sapphire-arches, 2024-07-27:
We should check to makes sure that we poll the buttons often enough that 100ms works as expected (so at least 20Hz)

m-m-adams, 2024-09-02:
I'm gonna close this with the vague hope that it got fixed by the scheduler rejigging

closed 2024-09-02 by m-m-adams