SynthstromAudible/DelugeFirmware #2317: Song view broken

deepblueday, 2024-07-21:
### Please describe the problem:

Recently bought a used Deluge. It came with a "nightly" build of community firmware 1.1.x.

Song view was unusable. The transport white bar that travels from left to right when playing did not display. The clips in the rows would not show up either, and in general the pads would have weird behavior. 

I reverted to the "Release" 1.1 community firmware, that did not fix the issue. 

I then reverted to the latest "official" synthstrom firmware, that DID solve the issue.

Any thoughts on what it could be? I'd rather use the community firmware with all its additional features.

Thanks!

### What is the expected behavior?

Song view works as expected, transport white bar displays, tracks light up when clips are available.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.1/Beethoven

### What is the firmware name:

deluge-c1_1_0.bin

### If possible provide the steps to reproduce the issue and upload additional media:

Turn on,  go to song mode, tap an empty pad to create a new clip, add some notes. Repeat to add a new track. The track does not show up in the song mode with colored clips. Tap play, the transport white bar does not display.

sapphire-arches, 2024-07-21:
sounds like the device has grid view set as default? Try holding the "song" button and turning select.

sapphire-arches, 2024-07-23:
nice. you can change the default in the settings menu as well.

closed 2024-07-23 by sapphire-arches