SynthstromAudible/DelugeFirmware #2325: MIDI CC Automation view doesn't work

wargoth, 2024-07-22:
### Please describe the problem:

When I record MIDI CC parameter changes they playback correctly, however:
1. when I navigate to automation view it's displayed as if the parameter is not automated at all. 
2. When I set gold knob to automated MIDI CC value, the display doesn't indicate it has been automated, but the value LEDs change as expected during playback

### What is the expected behavior?

1. Expect automation view to reflect recorded value changes.
2. When gold encoder is set to automated MIDI CC, expect display to indicate the MIDI CC is automated (indicated as dot in 7SEG display)

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Release 1.1/Beethoven

### If possible provide the steps to reproduce the issue and upload additional media:

1. Create MIDI track, use learn+audition button to learn MIDI input
2. Record MIDI CC value changes and make sure automated parameter plays back correctly
3. Go to automation view and use encoder to navigate to automated MIDI CC. Note automation view is not reflecting recorded value changes
4. Use gold knob encoder to set it to automated MIDI CC. Note there is no indication the parameter is automated

m-m-adams, 2024-07-27:
I can't reproduce so might need more info

What controller are you using, and have you checked the mpe settings on deluge? If deluge thinks the automation is coming from an mpe member channel it won't display it in the automation view (it displays mpe master channel automation)

wargoth, 2024-07-27:
Thank you for looking into this. I just removed MIDIFollow.xml which was created by the beta version and it started working as expected. Attaching the XML file for reference. [MIDIFollow.zip](https://github.com/user-attachments/files/16401478/MIDIFollow.zip)

closed 2024-07-27 by wargoth