SynthstromAudible/DelugeFirmware #2335: Grid mode playback skips rows.

XrwrX, 2024-07-26:
**Please describe the problem:**

In Grid mode playback (green mode), rows are not always played in the expected order. Starting playback on the top row might result skipped rows. For example, it can play row 1, skip row 2, and then play row 3. 

**Is there a relevant Pull request?**

No

**What hardware did you reproduce it with?**

OLED

**What firmware did you reproduce it with?**

1.2 Nightly 20240726

**If possible provide the steps to reproduce the issue and upload additional media:**

I don't have specific steps to reproduce, but we think it's related to Section colors from the row view. I personally experienced the issue and I was asked to open this bug by "ilovesteak" on Discord #support.

m-m-adams, 2024-07-27:
The order is determined by rows view, it's to accommodate reordering the sections

Grid should either allow reordering and always display in the order they'll play in, or force playback to follow the order in the grid

I'm not sure which is better and being unfamiliar with the grid code I think it's up to whoever fixes it

m-m-adams, 2024-08-08:
Fixed by #2400

closed 2024-08-08 by m-m-adams