SynthstromAudible/DelugeFirmware #2414: Follow up changes for PR #1545 (Song Macros)

seangoodvibes, 2024-08-09:
### Please describe the problem:

- grid mode colour is same as performance view and it's now oddly placed below where the yellow mode was
  - Solution: move it up and change its colour to yellow
- can't see clip launch playhead and bars / beats remaining when launching from within clip / keyboard view
- Pressing the same clip multiple time cycles though different modes - we should show those modes on the display as a pop-up. And the instructions should explain what colours correspond to what mode.

### What is the expected behavior?

See above

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

bfredl, 2024-08-09:
>    Solution: move it up and change its colour to yellow

I thought about this, but as the YELLOW mode was not actually removed from the enum, I thought it was reserved for future use.

seangoodvibes, 2024-08-09:
> >    Solution: move it up and change its colour to yellow
> 
> 
> 
> I thought about this, but as the YELLOW mode was not actually removed from the enum, I thought it was reserved for future use.

Ah that was just a mistake

Yellow mode was just removed recently so I prob forgot that part