SynthstromAudible/DelugeFirmware #2419: Allow users to exit Velocity View by hitting audition pad + velocity shortcut

ok-reza, 2024-08-10:
### Please describe the problem:

Holding an audition pad + velocity once in velocity view doesn't exit velocity view, it just adds a note at that shortcut.

### What is the expected behavior?

I think it should exit the view the same way we enter it.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

1.2

m-m-adams, 2024-08-11:
Do other shortcuts work like that?

seangoodvibes, 2024-08-12:
> Do other shortcuts work like that?

no, if you enter LPF Frequency menu with Audition + Shortcut you can't exit the menu using Audition + Shortcut. Only way to exit menu is with "Back" or Pressing grid

ok-reza, 2024-08-12:
i think my gut reaction to this desire is based on how automation view is implemented: you hit clip to enter automation view, and can always hit clip again to exit.

i was under the impression that velocity view was a separate view from automation view, so if audition+velocity is how you enter, i thought the same would get me out of it, like hitting clip for automation view.

i can see now that velocity view is integrated with automation view (it's included in the automation view overview), so you can hit the blinking clip pad to exit or hit back.

if audition pad+velcoity opens velcoity view, would it make any sense to let shift+velocity open the current velocity value that new notes would utilize (technically a new menu)? this would create a parity with how users enter automation view for other parameters, the only difference is this one is a mod source button technically...