SynthstromAudible/DelugeFirmware #2422: Can't delete a single kit row in C1.1.0

MkUltra85, 2024-08-11:
### Please describe the problem:

I'm trying to delete a kit row from a kit; if I hold the audition pad for a kit row and press the SAVE/DELETE button, it brings up the "SAVE SYNTH" menu, but does not delete the kit row. This is of course in a KIT and not a SYNTH. This seems to happen regardless of anything else I do.

### What is the expected behavior?

Delete the kit row

### Is there a relevant Pull request?

I don't believe so

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.1/Beethoven

### What is the firmware name:

C1.1.0

m-m-adams, 2024-08-11:
Not  a bug, the command to delete a kit row is to hold a pad in the row and press delete

m-m-adams, 2024-08-11:
audition+save is to save the kit row as a synth for later use

closed 2024-08-11 by m-m-adams

MkUltra85, 2024-08-11:
Hah, oops. I would swear I tried that, too, but it's working now.