SynthstromAudible/DelugeFirmware #2437: Two issues with the stem recording

thesoundfarmer, 2024-08-14:
### Please describe the problem:

Deluge seems to freeze when exporting stems on older tracks but the stem recording can be cancelled so the Deluge hasn't crashed.  Just hangs on 0/11 stems.

A new tune I am trying to export stems only exports some of the instruments and not others.

### What is the expected behavior?

Had hoped it would export all instruments

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

deluge-v1_2_0-nightly+2024_08_12-31d29ed

### If possible provide the steps to reproduce the issue and upload additional media:

Issue 1:  

Turned on
Loaded track
Pressed Save and Rec
Message reads "Exporting 0/11"
Fails to proceed

Issue 2:

Turned on
Pressed Save and Rec
Stems export happily
Open on computer
Only some of the instruments have saved.  Files are there but empty

thesoundfarmer, 2024-08-14:
I should add Issue 1 is exporting in clip mode
Issue 2 is exporting in song mode

m-m-adams, 2024-08-15:
I suspect 1 might be related to empty clips, they might start linear recording with auto length expansion

thesoundfarmer, 2024-08-15:
Hi,

The only muted instruments are the duplicates doing different parts of the melodies.  I'd have thought they'd unmute when they're cued

seangoodvibes, 2024-08-15:
> Hi,
> 
> The only muted instruments are the duplicates doing different parts of the melodies. I'd have thought they'd unmute when they're cued

Are you in arranger view?

In arranger view stem export will exclude muted tracks, if you want to export all tracks, unmute them all before stating stem export

If you're doing it in song view, it will export all clips regardless of mute state.

thesoundfarmer, 2024-08-15:
I am in arranger view.  All tracks in arranger view are unmuted.  When I press Shift and Rec, it automatically mutes all bar the first track and proceeds to export down the grid unmuting track by track

Once complete I have a folder with 10 out of the 12 tracks that make up the song.  9 of which are 5 seconds long and 1 which is the full length of the track.  2 of those 5 second tracks have sound, the other 7 are silent.

I'll record a video to illustrate once my phone has charged!

thesoundfarmer, 2024-08-15:
I've just tried exporting a different track in arranger view and a similar thing has happened.
All tracks 5 seconds long in this case, 49mb in size and only 2 with any audio

thesoundfarmer, 2024-08-16:
New issue discovered

Stem export in song and grid view - not all instruments exported.

thesoundfarmer, 2024-08-16:
To summarise and better explain the issues I'm having with stem export

Issue 1:  Exporting in both Arranger and Clip/Song view on particular songs
Deluge seems to freeze when exporting stems on older tracks but the stem recording can be cancelled so the Deluge hasn't crashed. Just hangs on 0/11 stems.

Issue 2:  Exporting in Arranger view
Most but not all stems export
Those that do are all the same file size (49mb) however some are only 5 seconds long whilst some are full length but silent
Others export fine

Issue 3:  Exporting in song and clip view
Most but not all stems export

seangoodvibes, 2024-08-16:
@thesoundfarmer can you collect all and save the song you're having all these issues with? 

Can you also confirm that this problem only happens with this song / older songs and not with new songs?

Thanks

thesoundfarmer, 2024-08-16:
Hi Sean,

Sure I’ll do that in a bit.  

Issue 1 is happening to an old track
Issues 2 and 3 are with a new one

Thanks!

seangoodvibes, 2024-08-16:
> Hi Sean,
> 
> Sure I’ll do that in a bit.
> 
> Issue 1 is happening to an old track Issues 2 and 3 are with new ones
> 
> Thanks!

Thanks

One more test I'd like you to try to confirm something.

If you go into the Song menu (press select from Song view or Arranger View) and go to the Stem Export menu, go into the Configure Export submenu.

Can you confirm if you are exporting with Normalization toggled ON or OFF?

If you have been testing with Normalization OFF, can you try with Normalization ON to see if it makes any difference?

Thanks

thesoundfarmer, 2024-08-16:
Hi,

Normalisation was off.  I turned it on but it still refused to export anything.
I tried to export in arranger too and still no luck

seangoodvibes, 2024-08-16:
> Hi,
> 
> Normalisation was off. I turned it on but it still refused to export anything. I tried to export in arranger too and still no luck

Can you share your problematic songs? (collect all save please)

Thanks

thesoundfarmer, 2024-08-16:
Here's a link to the files

https://drive.google.com/drive/folders/1PvdvVg4uJaVAvPYe83iDBGYBMG82L2_H?usp=drive_link

seangoodvibes, 2024-08-16:
@thesoundfarmer can you confirm if you experience this issue as well in the song you have issues with exporting from arranger?

https://github.com/SynthstromAudible/DelugeFirmware/issues/2455

thesoundfarmer, 2024-08-16:
In Prog21 I am unable to export in both arranger and clip/song at all

In DX7 I am able to export in both but there are instruments missing in both modes

thesoundfarmer, 2024-08-16:
I can play everything through in arranger mode without any issues

seangoodvibes, 2024-08-16:
@thesoundfarmer so a bit of an update:

re: the prog song, I noticed that exporting the first track doesn't finish because if you turn the VU meter on you'll see that the levels never quite drop to silence (it gets stuck with 3 green pads showing). The stem export tries to export until silence. So if it never reaches silence it will never finish exporting.

so what we need to figure out is:

1) why do the levels never drop to silence (even tho it's audibly silent) - this could be a bug with the way we measure levels in the deluge
2) make stem export configurable so that we can turn off exporting to silence

re: the dx7 song, I went to song view and was able to export all the clips, I saved them here if you want to take a look:

https://drive.google.com/file/d/123pGgIFBJCd90v7LrSrX2A30cb6eL4Oj/view?usp=share_link

seangoodvibes, 2024-08-17:
@thesoundfarmer I made an updated build that allows you to turn off "exporting stems to silence" in the stem export configuration menu

you can try that build by download the release build here: https://github.com/SynthstromAudible/DelugeFirmware/actions/runs/10427636367?pr=2456

Please let me know if this resolves all your issues with exporting.

This won't exporting tails unfortunately but will try to resolve the silence issue asap.

thesoundfarmer, 2024-08-17:
Many thanks for everything so far!

I've had another go exporting with the release build and "Exporting Stems to Silence" switched off

I can now export stems in clip mode with Prog21 which is great, however there is some audio bleed at the start of most of the clips.  

When I tried exporting from DX7 again it provides the message "Exported # of 46 clips" but only 40 to 42 of them actually export.  It's different every attempt.  There should be an organ in there but it doesn't seem to want to export.

Exporting in arranger view is still giving me mostly 5 second files which are either silent or have a bit of music on.

I'm now also receiving "SD Card Error" messages after each clip is exported so I'm going to try a different one.  The files have actually exported however

Thanks again!

seangoodvibes, 2024-08-17:
> Many thanks for everything so far!
>
> I've had another go exporting with the release build and "Exporting Stems to Silence" switched off
>
> I can now export stems in clip mode with Prog21 which is [...]

Hrm.

I think the audio bleed is because your clips have tails that are still ringing out. 

exporting to silence would help fix that but until we resolve why the tails don't go fully silent we can't use that.

so I'm now thinking we need a way to hard cut the audio before the next export happens.

seangoodvibes, 2024-08-17:
@thesoundfarmer I managed to get the organ to export. I had to rename the track from <<ORGAN>> to just Organ. Can you try that? I think the exporting doesn't like those << >> characters

thesoundfarmer, 2024-08-17:
Aah brilliant, I'll give renaming a go!

seangoodvibes, 2024-08-24:
@thesoundfarmer can you confirm with the latest nightly build if we can close this issue now? Thanks!

thesoundfarmer, 2024-08-25:
Hi Sean,

I’ve not had a chance to have too much of a play but from what I can see yes!  

The addition of a no fx option is brilliant 

Thanks again!

seangoodvibes, 2024-08-25:
Thanks for reporting back. I'll close this issue for now. If you notice the issue again, feel free to re-open this issue or start a new one. Thanks!

closed 2024-08-25 by seangoodvibes