SynthstromAudible/DelugeFirmware #2455: Arranger mode wont play

mcabrerapf, 2024-08-16:
### Please describe the problem:

In arranger view certain songs that I have saved have stopped being able to play the arrangment. If I press play the song tries to start but immediatly stops.

### What is the expected behavior?

Arranger to play the arranged song without issues

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

deluge-v1_2_0-nightly+2024_08_16-a0f4498

### If possible provide the steps to reproduce the issue and upload additional media:

Download the following song: https://drive.google.com/drive/folders/1xA_smRc8DcT4AknBBPjcnPII-_c7GhLK
- Turn deluge on
- Load song (should load into arranger view)
- The song will try to play but will stop immediatly as such: 

https://github.com/user-attachments/assets/0286293f-b7d6-4899-959f-2e88f08a6b26

m-m-adams, 2024-08-17:
Could this be a midi issue? If clock ends up looping back into the deluge this is a possible symptom

mcabrerapf, 2024-08-18:
> Could this be a midi issue? If clock ends up looping back into the deluge this is a possible symptom

That was my first guess but I have midi feedback disabled and nothing is plugged into the deluge in the video.
Also the song view plays correctly and some other arrangements work.

mcabrerapf, 2024-08-20:
I can confirm now that the nightly build `deluge-v1_2_0-nightly+2024_08_06-b2e7f35` is where the issue starts showing up.
The last nightly where it was working is this one `deluge-v1_2_0-nightly+2024_08_03-a78b984`

seangoodvibes, 2024-08-20:
Confirmed that this bug was introduced by the tempo automation PR #2367

m-m-adams, 2024-08-21:
https://github.com/user-attachments/assets/3c94b0eb-480e-403b-bd12-214d9bf106c2

Can't reproduce, song plays on current nightly

closed 2024-08-21 by seangoodvibes