SynthstromAudible/DelugeFirmware #2465: Freeverb Bug

seangoodvibes, 2024-08-20:
### Please describe the problem:

There is an issue with the Freeverb model where even after turning playback off, the VU meter level does not drop to zero.

### What is the expected behavior?

Turning playback off should eventually drop VU meters to zero.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly, Built from community

seangoodvibes, 2024-08-27:
See prog song from https://github.com/SynthstromAudible/DelugeFirmware/issues/2437 to reproduce the free verb bug

m-m-adams, 2024-10-10:
I'm closing this since it's not a fixable bug, it's just how freeverb works