SynthstromAudible/DelugeFirmware #2476: Error M000 when exiting

alifeinbinary, 2024-08-22:
### Please describe the problem:

System crashes with Error M000 when scrolling in the Settings > MIDI > Devices submenu. 

### What is the expected behavior?

To highlight an item from the list [DIN, Loopback, $device]. Three items can fit on the OLED screen, however, when scrolling with the Select knob the bottom item (MIO in my case) will get nudged to its own page after which point a crash will occur. 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

other

### What is the firmware name:

1.2.0-DEV-2EBE56FA

### If possible provide the steps to reproduce the issue and upload additional media:

1. Connect a USB MIDI interface to the Deluge's host port so that three items populate in the Settings > MIDI > Devices submenu. I'm using the iConnectivity Mio. 
2. In a fresh project, enter the Settings > MIDI > Devices and use the Select knob to scroll up and down. 
3. The bottom item will get nudged to its own page and a crash will occur 
[image]

m-m-adams, 2024-08-24:
Are you using the deluge as a host or a device?

alifeinbinary, 2024-08-24:
Yes @m-m-adams , the Deluge is hosting the 1x1 Mio interface via it's USB port with the conversion adapter. No USB in between, just a direct connection to the Mio and the Mio is connected to DIN on the Osmose (fwiw). 

I've built from commit 4FF0CBBF to see if #2478 fixed the issue but, alas, it persists! @nikodemus was your fix meant to address #2476 or some other OLED scrolling glitch?

m-m-adams, 2024-08-25:
Nah that one is an animation fix for the stuttering that sometimes happens

seangoodvibes, 2024-08-25:
> Yes @m-m-adams , the Deluge is hosting the 1x1 Mio interface via it's USB port with the conversion adapter. No USB hub in between, just a direct connection to the Mio and the Mio is connected to DIN on the Osmose (fwiw).
> 
> I've built from commit 4FF0CBBF to see if #2478 fixed the issue but, alas, it persists! @nikodemus was your fix meant to address #2476 or some other OLED scrolling glitch?

There is however another PR that changed how scrolling happens in the menu so that's probably the culprit. I'll try to find it.

seangoodvibes, 2024-08-25:
It's either this one: https://github.com/SynthstromAudible/DelugeFirmware/pull/2450

or this one: https://github.com/SynthstromAudible/DelugeFirmware/pull/2451

PR #2592 (2024-09-08): Bugfix/usb midi dropped messages
Fixes some bugs around hosting multiple devices. The menu still doesn't look right (scrolling up gets weird) but it no longer crashes. Some issues with hubs and packing multiple midi messages into one USB packet are fixed by lowering our max size significantly, but that doesn't affect anything other than hubs and this seems to work fine.

I think the real downside would come from sending big sysex packets to a hosted device. We don't have any way to do that at the moment so it can be a future problem.

@bfredl fyi in case this impacts the sysex stuff

fix #2580 [... cut]
PR #2616 (2024-09-09): Bugfix/usb midi dropped messages
Fixes some bugs around hosting multiple devices. The menu still doesn't look right (scrolling up gets weird) but it no longer crashes. Some issues with hubs and packing multiple midi messages into one USB packet are fixed by lowering our max size significantly, but that doesn't affect anything other than hubs and this seems to work fine.

I think the real downside would come from sending big sysex packets to a hosted device. We don't have any way to do that at the moment so it can be a future problem.

@bfredl fyi in case this impacts the sysex stuff

fix #2580 [... cut]
closed 2024-09-08 by m-m-adams