SynthstromAudible/DelugeFirmware #2483: Lag on RATIO and ATTACK on per Instrument Compressor

racecube, 2024-08-24:
### Please describe the problem:

when I try to edit RATIO in a synth compressor (or in a single instrument of a clip) it is very slow in changing values. 
ATTACK also...but only from 0 to 20, from 20 up it works  normal.
Master Compressor has not that behaviour. 

### What is the expected behavior?

Quick and seamless changes

### Is there a relevant Pull request?

dont know

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.1/Beethoven

### What is the firmware name:

C1.1.0

racecube, 2024-08-24:
Explanation by REZA:

the ratio and attack stuff is because the values are implemented on a curve, it's more obvious it could be implemented better when you are in a clip compressor menu because each notch turn of doesn't necessarily mean a new value will be updated. originally, this curve was designed specifically for the gold knobs, where it feels a little more natural. i think a possible solution is having more decimal values for the first couple ratio values eg 1:1, 1.2:1, 1.4:1, etc and for attack levels

m-m-adams, 2024-08-24:
Purely UI, it's adjusting on every tick underneath

closed 2024-08-31 by seangoodvibes