SynthstromAudible/DelugeFirmware #2513: Freeze upon audio recording in Arranger mode

flipper70, 2024-08-28:
### Please describe the problem:

In newer nightly/beta builds (20240828 back to at least 20240826) the deluge is freezing when trying to record audio in arranger mode. Audio recording to the same tracks in song/grid mode is still working. 

### What is the expected behavior?

Recording into an audio track in arranger mode should work without deluge freezing.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly, Beta

### What is the firmware name:

1.2.0-BETA-1BF601B

### If possible provide the steps to reproduce the issue and upload additional media:

Start with new song
go into arranger mode
create an audio track
press record, press play

seangoodvibes, 2024-08-28:
This is probably due to the refactoring I did in arranger view

I'll check into this

seangoodvibes, 2024-08-29:
@flipper70 can you film a video of what you're doing? I'm trying your steps and I don't crash. Thanks

PR #2526 (2024-08-31): Bugfix/arranger recording
Fix #2513 

Fixes a loop that could be infinite 

Change audio looping to avoid setting next tick to 0 when not necessary
PR #2529 (2024-08-31): Bugfix/arranger recording
Fix #2513 

Fixes a loop that could be infinite 

Change audio looping to avoid setting next tick to 0 when not necessary
closed 2024-08-31 by seangoodvibes