SynthstromAudible/DelugeFirmware #2547: Oled screen issue

TwoToneProductions, 2024-09-01:
### Please describe the problem:

Intermittently my oled screen shows nothing or the text is on the far right 

other times it works perfectly fine 

### What is the expected behavior?

no the screen in blank

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly, Beta

### What is the firmware name:

deluge-v1_2_0-beta+2024_09_01-e1862cc.bin

### If possible provide the steps to reproduce the issue and upload additional media:

I turned it on and the screen bugs out occasionally - only has done this since t=ive used the new update I can use the nightly update at all only the Choppin update

m-m-adams, 2024-09-01:
Sorry could you clarify which firmwares are causing you issues?

TwoToneProductions, 2024-09-01:
Hi 

I had this problem with the following Bin files

deluge-v1_2_0-beta+2024_09_01-e1862cc.bin - with this one its intermittant

deluge-v1_2_0-nightly+2024_09_01-220ea50.bin - with this one I couldn’t get it to work

I hope this helps 

Amazing work guys 

Kind regards

Paul Carter

> 
> 
> Sorry could you clarify which firmwares are causing you issues?
deluge-v1_2_0-nightly+2024_09_01-220ea50.bin

TwoToneProductions, 2024-09-01:
Also to double check it wasn’t a sd card issue I went back to my old bin file and is now working perfectly.Sent from my iPhone

TwoToneProductions, 2024-09-02:
Hi Mark,

Further update the stock software is showing up with the same issues

Not sure if this is a coincidence or problems with oled that was fitted.

However can confirm I I zero issues with my screen previous to the the update.

Just to be safe Ive logged it with Synthstrom as a hardware defect.

Ill leave you guys to decide if this is linked or just a hardware problem.

Kind regards

Paul Carter

m-m-adams, 2024-09-02:
Ah ok thanks!

Closing this for now but please reopen if it turns out to be a software problem

closed 2024-09-02 by m-m-adams

TwoToneProductions, 2024-09-02:
No problem mark,From research it looks like the issue is with the boot loader.How do I ensure I have the right version on my oled apparently this has issues if not done correctly.Thanks Paul Sent from my iPhone