SynthstromAudible/DelugeFirmware #2558: New clip naming ability blocks renaming the synth preset without saving.

ok-reza, 2024-09-02:
### Please describe the problem:

The new clip naming ability allows the user to make a secondary name for the clip (eg. chorus, verse, lead, etc.) that appears below the preset name.

However, now the user cannot change the name of the primary clip name, unless they save it with the desired name. This creates unnecessary duplicate presets just so the user can have the desired name display.

### What is the expected behavior?

Before, SHIFT+NAME in a clip would allow the user to change the preset name of the clip, without the need to save the preset. This allowed for quick organization without populating the preset list. This should be brought back as editing both the primary and secondary name of a clip is equally important.

I suggest SHIFT+NAME open a menu with 1 of potentially 3 options:
- Change Instrument Name (changes name for instrument and all clips utilizing that instrument)
- Change Clip Name (changes the name of just the current clip)
- Change Kit Row Name (if a user is in a kit clip, this would change the name of the current active kit row when the user entered the shortcut. this option wouldn't show up in non-kit clips).

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

### What is the firmware name:

1.2 from 8/31

m-m-adams, 2024-09-08:
@KrisGelero I think this is from your naming PR, are we sure what the previous behaviour was?

KrisGelero, 2024-09-10:
@m-m-adams 
This is more a feature request. Naming presets was only possible through (from the official manual) ; 

[image]

I tried this also on official, and it was not possible with SHIFT+NAME to name anything. Only in the KIT it worked to changed the drum sound name.

seangoodvibes, 2024-09-10:
This will be taken care of in the next release by the naming menu I'm working on.

If you couldn't rename synth track from in a synth clip previously then we're good I think.

closed 2024-10-01 by m-m-adams