SynthstromAudible/DelugeFirmware #2573: Stem Exporting again

thesoundfarmer, 2024-09-03:
### Please describe the problem:

Hi, me again

I am still having trouble with stem exporting.

I'm in arranger view wanting to export the 12 instruments that make my up track.

Normalise is off (I've tried with on too)
Export to silence is off
FX are off

I press save and record and the Deluge exports.  12 files appear on the sd card as the same file size (in this case 64mb). 
 However when I import them into Reaper or ProTools, the majority of them are 5 seconds long.  Occasionally one is as long as the track should be, the rest are still 5 seconds long and silent.

Nothing is muted in Arranger view until each track is processed.

### What is the expected behavior?

To be able to fully export all the instruments in arranger  view

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

deluge-v1_2_0-nightly+2024_09_03-36b2b72

### If possible provide the steps to reproduce the issue and upload additional media:

[image]

[image]

seangoodvibes, 2024-09-03:
@thesoundfarmer can you please upload a collect all version of your song please so I can try to reproduce your export issue? Thanks

thesoundfarmer, 2024-09-04:
Sadly I can’t :( 

I have an error message saying “You’ve encountered an unspecified error.  Please report this bug”

[image]

thesoundfarmer, 2024-09-04:
It happens on every track I’ve tried to export from arranger

seangoodvibes, 2024-09-04:
> Sadly I can’t :(
> 
> I have an error message saying “You’ve encountered an unspecified error. Please report this bug”
> 
> ![IMG_4103](https://private-user-images.githubusercontent.com/25672006/364259373-6e499dc7-9baa-4110-b14a-6cf2fcf9cc5a.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjU0NTg4MTYsIm5iZiI6MTcyNTQ1ODUxNiwicGF0aCI6Ii8yNTY3MjAwNi8zNjQyNTkzNzMtNmU0OTlkYzctOWJhYS00MTEwLWIxNGEtNmNmMmZjZjljYzVhLmpwZWc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTA0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkwNFQxNDAxNTZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xMTU3YmNmMDkwODE1MTZhOTNhNjQ2MTg5YzE0Y2JkZTMyMWM2Zjc0MzNjZGZjODQ2M2I3MDUzY2UwZmFiYTBjJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.BiGxu38zkuXKnCH9IJ6ZcczzxTVYE-sfOfEPQKVibzU)

Hey, I think there's confusion.

When I said collect all and save I meant:

> press save, then press and hold select, it'll save the whole project into a folder 

Thanks!

thesoundfarmer, 2024-09-04:
That’s what I did, gave me that error message 

Greg

seangoodvibes, 2024-09-04:
> That’s what I did, gave me that error message
> 
> Greg

ah shoot, ok thanks for letting me know

seangoodvibes, 2024-09-04:
If you're using the nightly can you try collect all and saving with the beta? Thanks

thesoundfarmer, 2024-09-04:
I’m using the Nightly,

Sure I’ll try it with the beta later on and send it over!

thank you!

greg

seangoodvibes, 2024-09-04:
@thesoundfarmer have you tried re-loading the "5 second" stems on the deluge? Another person with this issue mentioned that the stems load fine on the deluge and are the right length, etc.

Which leads me to believe it's something with how the wav files are read by external software.

thesoundfarmer, 2024-09-04:
Looks like I can indeed open them in the Deluge!

feilsch-hering, 2024-09-05:
Hello Sean, as discussed previously in another treat, I have the exact same problem with the stems only containing 5 seconds of audio. I too get the same error message when I try to save with collect all media, and I tried all the suggested settings already.
I Tried to upload one of the stems for you here but that failed due to the 25mb size limit. Like suggested earlier, we-transfer is a better alternative, so here is the downloadlink: https://we.tl/t-Xt2Z0PkKmH
Anything else I could try?

seangoodvibes, 2024-09-05:
@thesoundfarmer @feilsch-hering 

Can you try collect all and save using the beta build instead of the nightly build? The nightly build has a bug with saving.

Thanks

feilsch-hering, 2024-09-06:
Hi Sean, here is my track saved with collect media, I hope it contains everything you need.
[Fungtek-0-1 15Daniel Fritschi.zip](https://github.com/user-attachments/files/16907139/Fungtek-0-1.15Daniel.Fritschi.zip)
best

phpguru, 2024-09-07:
Don't mean to hijack this thread but is Jan 1 1970 dates on exported stems also a bug?

seangoodvibes, 2024-09-07:
> Don't mean to hijack this thread but is Jan 1 1970 dates on exported stems also a bug?

Nah not a bug. All files on the deluge export with that date because the deluge doesn't know the current date.

Not ideal, but not entirely sure how we would resolve that tbh.

seangoodvibes, 2024-09-11:
@feilsch-hering @thesoundfarmer while not an ideal solution, we recommend the following to allow you to export stems:

## Troubleshooting

### I have a track that won't export

#### Scenario: Track(s) get exported with only 5 seconds of audio

Several users reported that their arranger stems were not getting exported properly as they included only 5 seconds of audio.

All of these users were using offline rendering.

All of these users had complex arrangements.

Solution:

For heavy arrangements, if you encounter the above issue, we recommend turning off `Offline Rendering` in the `SONG\EXPORT STEMS\CONFIGURE EXPORT\` menu.

The problem is due to memory filling up faster when using `Offline Rendering` compared to `Online (Live) Rendering`. We hope to find a solution for this problem as soon as possible.

thesoundfarmer, 2024-09-12:
Hi Sean,

Sorry I've not been able to help out since my initial post, been mad busy the past couple of week, barely any time for the Deluge.

The solution worked a treat so thanks very much again!

feilsch-hering, 2024-09-13:
Hi Sean, I too can confirm that I now managed to create proper wave stems, when I disabled the Offline Render option. Thank you for pointing that out. I thought I tried all options previously already without fixing the issue, but maybe I was wrong and in fact missed to try this option.

seangoodvibes, 2024-09-13:
Great to hear @thesoundfarmer and @feilsch-hering that this worked!

I actually added this option recently so it wasn't there before when you were testing out the various options.

feilsch-hering, 2024-09-14:
Ah, OK then thank you for providing this new option! I already thought it didn't look familiar.

PR #2667 (2024-09-17): Fix offline rendering in memory limited songs
Finishes all reads/writes before continuing to render during offline rendering

Fix #2573
PR #2668 (2024-09-17): Fix offline rendering in memory limited songs
Finishes all reads/writes before continuing to render during offline rendering

Fix #2573
closed 2024-09-17 by seangoodvibes