SynthstromAudible/DelugeFirmware #2629: Pressing any pad while the Deluge logo animates/is visible during boot up will freeze the device.

ok-reza, 2024-09-09:
### Please describe the problem:

Perhaps only OLED?
While the Deluge boots up and the logo animates and holds, if you press ANY pad after turning the device on, the deluge will freeze and hang with the logo visible on the display.

### What is the expected behavior?

Pressing any pad after the device is turned on should have no effect.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

### What is the firmware name:

1.2-beta

seangoodvibes, 2024-09-10:
I also experienced this holding the play button while booting up (wanted to flash firmware and pressed play instead of shift lol)

mweigt, 2024-09-30:
Unfortunately, as of
[4b3b6d6](https://github.com/SynthstromAudible/DelugeFirmware/commit/4b3b6d65e772dafe5365c8cad48e93de43d2cb7b)

I can still crash/hang my OLED Deluge by pressing random buttons/pads during boot, even though #2611 was reversed by #2642. 
(Pressing the select encoder during boot to reset settings also does not work, which is probably related)

So, while I am happy that this was not my fault, the issue should be reopened...

seangoodvibes, 2024-09-30:
> Unfortunately, as of [4b3b6d6](https://github.com/SynthstromAudible/DelugeFirmware/commit/4b3b6d65e772dafe5365c8cad48e93de43d2cb7b)
>
> I can still crash/hang my OLED Deluge by pressing random [...]

re-opened, thanks for flagging this

PR #2967 (2024-11-24): Bugfix/freeze_while_booting
ui hasn't loaded  when the busy wait routine is first called yet if the user presses a pad while the flash is still being accessed

Fix #2629
PR #2968 (2024-11-24): Bugfix/freeze_while_booting
ui hasn't loaded  when the busy wait routine is first called yet if the user presses a pad while the flash is still being accessed

Fix #2629
closed 2024-11-24 by stellar-aria