SynthstromAudible/DelugeFirmware #2635: MIDI MAPPING LOST WHEN CHANGIN PRESET

bukru, 2024-09-10:
### Please describe the problem:

I tried to build a config with deluge and oxygen49 midi keyboard.
I did an init template with 4 voices (1 bass, 1 drum kit, 2 more synth sound).
I mapped everything so i have voice's volume on faders, and i can switch instrument by changing global midi channel on the keyboard.

But ... dramma : when i change preset instrument on deluge all midi mapping of that voice is lost : (

Why ?

### What is the expected behavior?

Midi mapping should keep in place when changing song preset.

I understant that it could not be possible if i change type of voice (ex : synth to kit), but it could be possible when the type of voice is unchanged (kit to kit, synth to synth). 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.1/Beethoven

### If possible provide the steps to reproduce the issue and upload additional media:

- Attach midi keyboard with fader etc...
- create empty project.
- create 4 instruments. (kit or synth doesn't matter).
- midi map the master volume of each voice to faders.
- try it (should work).
- change song preset of some voice.
- try it (midi mapping is lost for that voice).

seangoodvibes, 2024-09-10:
This is how the deluge has always worked. It may be fixable but, this is also why I built midi follow as a feature so you didn't need to all the manual learning over and over. Setup your controller once and use it everywhere.

bukru, 2024-09-10:
I read that documentation : https://github.com/SynthstromAudible/DelugeFirmware/blob/community/docs/features/midi_follow_mode.md

I understand the behaviour : have a default midi mapping for instrument, active for the last active instrument/clip.
It's a good thing for sure and thanks a lot for that. But it does not allow to have parameters from multiple instrument mapped to the midi controller. (for exemple have all the master volumes mapped to the faders).
I consider using that feature.

I have a questions : 

- I read on the documentation "Note: Your existing MIDI learnings will remain untouched and can be used together with the master MIDI follow mode."
==> if i use this feature and map a parameter on a spécific control of my midi controller (ex : cutoff instrument 1 on Knob number 1), does it override the default mapping from the midi follow feature ?

- can i remove some lines on the xml file to have default mapping for less parameters ?

Thanks a lot for support this is awesome

seangoodvibes, 2024-09-10:
> I read that documentation : https://github.com/SynthstromAudible/DelugeFirmware/blob/community/docs/features/midi_follow_mode.md
>
>
>
> I understand the behaviour : have a default midi mapping for [...]

Midi learning will not override midi follow

They should both keep working

You can disable parameters from being used with midi follow by entering 255 in the xml file as explained here:

[image]

closed 2024-09-10 by bukru