SynthstromAudible/DelugeFirmware #2638: Stuck during boot after flashing v1_2_0-nightly+2024_09_10-4d6efd9

feilsch-hering, 2024-09-12:
### Please describe the problem:

I tried to flash deluge-v1_2_0-nightly+2024_09_10-4d6efd9, it seems to update without errors, but on reboot it takes unusually long displaying only a pattern on the display and finally after ~ 30 seconds some red pads light up and Deluge freezes. 
I Tried re flashing the nightly but with the same result. Gladly flashing deluge-v1_2_0-beta+2024_09_10-077dd8f restored the normal boot procedure and it works again.
[image]

### What is the expected behavior?

Normal boot.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Nightly

### What is the firmware name:

deluge-v1_2_0-nightly+2024_08_08-6fa9e2d

### If possible provide the steps to reproduce the issue and upload additional media:

1. Turned on, while holding shift, 2. flashing deluge-v1_2_0-nightly+2024_08_08-6fa9e2d

x3003, 2024-09-13:
Yes, that's an issue. Wanted to report it myself, but too late.
The deluge seems to boot ok [read: not stuck on the logo] when not having the card in the slot.

neilalbrock, 2024-09-13:
+1 I had the same issue

x3003, 2024-09-14:
v1.2.0 nightly 74bb6eb seems to have fixed the problem.

closed 2024-09-17 by m-m-adams