SynthstromAudible/DelugeFirmware #2658: Midi configurations in kit row aren't saved

bukru, 2024-09-15:
### Please describe the problem:

In a kit, press [AUDITION] pad + [Midi] to convert the row in midi row and then choosing midi channel and note with gold knobs.
Save the kit.

When reloading the kit the configuration is lost.

### What is the expected behavior?

The configuration should be saved with the kit.

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Release 1.1/Beethoven

### If possible provide the steps to reproduce the issue and upload additional media:

In a kit, 
-press [AUDITION] pad + [Midi] to convert the row in midi row
-then choosing midi channel and note with gold knobs.
-Save the kit.

- When reloading the kit the configuration is lost.

seangoodvibes, 2024-09-15:
Midi isn't saved with presets, it's saved with the song. This is a bigger issue.

bukru, 2024-09-16:
Thanks for answer,

Is it possible to copy/paste a clip and his midi configuration from a song to another ?

m-m-adams, 2024-09-17:
@seangoodvibes Midi drums are still saved in the kit and I think that's what's being described 

I can't reproduce this though. Midi rows in my kits are loading fine

seangoodvibes, 2024-09-17:
> @seangoodvibes Midi drums are still saved in the kit and I think that's what's being described
> 
> I can't reproduce this though. Midi rows in my kits are loading fine

ah yea that works fine for me too.

bukru, 2024-09-17:
Thanks to all of you as always for your support :+1:   

So i retry the stuff, and as you say me it work fine for you i searched what i did wrong and ... i think i found it !

When saving : 
- <save> + <kit>  (ok)
- select with <select> knob OR write a name with keyboard (ok)

- and then here is it :
     IF you write a new name then press "enter" with the keyboard layout, it saved the new kit
     **BUT IF you select a existing kit, then press "enter" with the keyboard layout, it will ask for "overwrite ?", then press "enter" again, and it will do nothing actually.** 

As a conclusion : i was thinking to save the kit, but actually not >__________<

Now i know it but i think i will do this error sometimes again.

bukru, 2024-09-17:
Do you think i should post an other issue for that or i just close the thread ?

seangoodvibes, 2024-09-17:
> Do you think i should post an other issue for that or i just close the thread ?

So if I'm understanding correctly, what you're saying is that when you press enter to overwrite an existing kit, it's not saving your midi changes into that overwritten kit. I can test that

bukru, 2024-09-17:
Actually i think it save nothing in that situation

bukru, 2024-09-19:
When saving :

    < save > + < kit > (ok)

    select with < select > knob OR write a name with keyboard (ok)

    and then here is it :
    IF you write a new name then press "enter" **with the keyboard layout** (the green "key" on the right), it saved the new kit

    **BUT IF** you select a existing kit, then press "enter" **with the keyboard layout  (the green "key" on the right)**, it will ask for "overwrite ?", then press "enter" again  **(the green "key" on the right)**, and it will do nothing actually, just close the popup

maybe i should made a video ?

m-m-adams, 2024-10-01:
Fixed by #2612

closed 2024-10-01 by m-m-adams