SynthstromAudible/DelugeFirmware #2660: Synth sounds very different (long release on one oscillator?) in 1.2 beta compared to 1.1

sinewave440hz, 2024-09-15:
### Please describe the problem:

I have a demo project which repros this. It consists of a bass synth and a drum machine. It sounds fine in Beethoven but in 1.2 the bass synth sounds completely different - as if the release of one of the oscillators is very long. I would suggest that the 1.2 firmware is interpreting the existing project differently somehow; I loaded the project that I created in Beethoven and saved it in 1.2 and then opened both projects and Beethoven again and they sounded identical. This is obviously a pretty serious issue that would mean that many of us couldn't safely upgrade, so do let me know if there's anything else I can test to pinpoint the issue.

### What is the expected behavior?

A project saved in Beethoven should sound exactly the same in 1.2. 

### Is there a relevant Pull request?

No, don't think so.

### What hardware did you reproduce it with?

7SEG

### What firmware did you reproduce it with?

Beta

### What is the firmware name:

1.2.0-beta-077dd8f

### If possible provide the steps to reproduce the issue and upload additional media:

1. In Beethoven , open attached project, note how the bass sounds
2. In 1.2, open same project, notice the difference in how the bass sounds

[Test1.zip](https://github.com/user-attachments/files/17007403/Test1.zip)

ok-reza, 2024-09-16:
confirmed that the release sounds longer in 1.2

as far as i can tell, all parameters are the same on both firmware. the env1 release is set to 37 in both, but raising the release in community 1.0.1 has much less effect than it does in 1.2, where 1.2 is already longer than 1.0.1's and can be made longer

if one holds down an audition pad, the filters/envelope seem the same.

i think there's something going on with the relationship between sustain and release being treated differently in 1.2

the only thing remotely similar to this is #2268, just mentioning that in case any breadcrumbs are to be found

m-m-adams, 2024-09-27:
> I loaded the project that I created in Beethoven and saved it in 1.2 and then opened both projects in Beethoven again and they sounded identical. This is obviously a pretty serious issue that would mean that many of us couldn't safely upgrade, so do let me know if there's anything else I can test to pinpoint the issue.

Sorry could you clarify this? Do you mean that after saving and reloading in beethoven it worked properly?

seangoodvibes, 2024-10-06:
> > I loaded the project that I created in Beethoven and saved it in 1.2 and then opened both projects in Beethoven again and they sounded identical. This is obviously a pretty serious issue that would mean that many of us couldn't safely upgrade, so do let me know if there's anything else I can test to pinpoint the issue.
> 
> 
> 
> Sorry could you clarify this? Do you mean that after saving and reloading in 1.2 it worked properly? 

I think they're saying that they opened the 1.1 project in 1.2, saved it, re opened it in 1.1 and it still sounded the same.

Which I guess confirm's 1.2's saving isn't causing the issue.

sinewave440hz, 2024-10-06:
@seangoodvibes yes, that's exactly what I am saying 👍 1.2's saving is definitely not the issue, because simply opening a project created in 1.1 causes the issue - no saving involved ie. it is how 1.2 is loading the existing data that is different. I have also noticed that the delays on some of my instruments have changed in that situation too. I've not yet established the behaviour there though...

seangoodvibes, 2024-10-26:
Linking this PR as the cause: https://github.com/SynthstromAudible/DelugeFirmware/pull/2004

seangoodvibes, 2024-12-07:
@sinewave440hz this bug should be fixed however I'm going to leave this issue open because our fix was a temporary one. We need a better fix for this. But your song should sound the same now.

m-m-adams, 2024-12-26:
I'm gonna close this, we should track the underlying issue seperately

closed 2024-12-26 by m-m-adams