SynthstromAudible/DelugeFirmware #2661: Kit sounds Filename changes in 1.2 beta from 1.1 release

synthcatlucas, 2024-09-16:
### Please describe the problem:

The file path seems to have changed and is making it so that samples in existing kits are not present. 

In this case SAMPLES/DRUMS/Undrgrnd/ is coming out as SAMPLES/DRUMS/UNDRGR~1/

In the screen shot provided If you look at the right above the blue folders you will find the folder name. Looking at visualcode line 59 you will see the file path changes to UNDRGR~1 from Undrgrnd. 
[image]

### What is the expected behavior?

File path should not be altered. Kits should load with sounds. 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

ok-reza, 2024-09-16:
have you used any file management apps, like deluge commander, or other applications to bulk adjust .xml files, sample names, folders, etc?

i would try and run a checkdisk type program to test the validity of your SD card, i wonder if it's getting corrupted, but the fact that the .xml is updated too makes me think that's not the case. how full is the SD card?

wouldn't be a bad time to backup the contents of your SD card if you haven't already while this is figured out btw.

m-m-adams, 2024-09-16:
Yeah this looks fairly unlikely to have been caused by the deluge. It's certainly not impossible but deluge will only write the names of files which it loaded successfully, so there must have been an UNDGR~1 folder at some point

closed 2024-10-01 by m-m-adams