SynthstromAudible/DelugeFirmware #2663: Hanging Notes through Midi Merge

requiets, 2024-09-16:
### Please describe the problem:

I am running a T-1 and Osmose through a Kenton 4 merge (DIN) connected to a few desktop synths (Nymphes and Minitaur) and am getting stuck notes pretty much immediately once playing. I've tried a few different patterns with the T-1 and switching the Osmose to poly aftertouch and single channel midi and am getting the same issues. I was actually getting this issue before using the powered midi merge (using each device individually at a time) in hopes the issue was a lack of power. Also happens using a Keystep 37. Only remedy to stop the stuck notes is to change the midi channel on Deluge.

So far I have tried a two week old version of Chopin, current build of Chopin and Beethoven, all with the same issue. 

### What is the expected behavior?

No stuck/hanging notes. 

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Beta

### What is the firmware name:

Chopin, Beethoven

### If possible provide the steps to reproduce the issue and upload additional media:

Turned on Deluge, Turned on Osmose, Set Osmose to Classic Keyboard (single channel midi), Learn Osmose, Set Deluge to Midi Channel 4 for Nymphes, Play Osmose, get stuck notes pretty much immediately, change midi channel to stop noise.

m-m-adams, 2024-09-16:
Can you make a midi recording using something other than the deluge and share it? This will be impossible to reproduce as we don't have the same gear as you

Could you also provide more detail on your setup? It's not clear how things are connected. A photo or diagram would help

Does the same thing happen on the official firmware?

requiets, 2024-09-17:
Here are a few things that are hopefully helpful. The setup is Osmose and T-1 Sequencer -> Midi Merge -> Deluge DIN In -> Deluge DIN Out -> Nymphes -> 16Rig/Logic (see video and audio). The audio file "No Deluge" is the same setup but bypassing Deluge all together.  I've had this happen on Beethoven and all versions of Chopin I've tried so far (today's, last week, week before). 

[image]
![IMG_8290](https://github.com/user-attachments/assets/d
[image]
aa690d6-112f-4fe0-8e9f-d7003688ea96)
![IMG_8292](https://github.com/user-attachments/assets/64ed
[image]
a2ce-4a66-4b10-b33c-9ce819adf355)
https://github.com/user-attachments/assets/02a7ebf5-2369-44db-ad6a-349162d0a4f2

[Audio.zip](https://github.com/user-attachments/files/17021275/Audio.zip)

requiets, 2024-09-17:
Looks like a there was an issue with a few of the uploads.

https://github.com/user-attachments/assets/01cd1237-50f1-404d-908e-e9f01d99dad8

[image]

m-m-adams, 2024-09-17:
Does removing the midi merge and connecting osmose->deluge->nymphes fix the problem?

m-m-adams, 2024-09-17:
Ok so just to clarify the chain of "osmose->deluge->nymphes" with no midi merge and no t-1 has problems? If so can I get more info since that I should be able to reproduce

requiets, 2024-09-17:
The midi merge is included in both the chain with and without the Deluge.
Only the chain involving the Deluge has problems. Swapping out the T-1 with
the Osmose in both chains has the same issue.

As heard in the audio, I can immediately create this issue in the Deluge
chain when a multiple notes are played together in quick succession using
the Osmose. Maybe Deluge isn’t sending midi off to Nymphes when too many
notes are being received from Osmose?

seangoodvibes, 2024-09-17:
> The midi merge is included in both the chain with and without the Deluge. Only the chain involving the Deluge has problems. Swapping out the T-1 with the Osmose in both chains has the same issue. As [...]

Can you remove the midi merge from the chain altogether and see if that works? That's what we want to know.

Osmose Midi Out -> Deluge Midi In
Deluge Midi Out -> Nymphes

requiets, 2024-09-17:
Yes, this happens when connecting without the midi merge. Maybe a bit less
frequent without the midi merge.

m-m-adams, 2024-09-22:
Could you make a midi recording of the osmose in a daw that I can play back to replicate the bug?

closed 2024-12-01 by m-m-adams