SynthstromAudible/DelugeFirmware #2693: Bug with "Copy Clips" in Grid View Green Mode

seangoodvibes, 2024-09-21:
### Please describe the problem:

Noticed that you can get stuck in UI_MODE_CLIP_PRESSED_IN_SONG_VIEW with "Copy Clips" pop-up stuck on the display

Steps to re-produce:
- Go into green mode in song grid view
- Press the grid to create a whole bunch of clips
- Try to create clips in multiple pads by pressing pads adjacent pads together
- You'll see that if you do it often enough you'll eventually get stuck with "copy clips" on the display

### What is the expected behavior?

Proper handling of multiple pad presses so that you don't get stuck in a UI mode

### What hardware did you reproduce it with?

OLED

### What firmware did you reproduce it with?

Built from community

m-m-adams, 2024-09-22:
repro is just press a second pad while the clip creation menu is still open on the first one

seangoodvibes, 2024-09-27:
> repro is just press a second pad while the clip creation menu is still open on the first one

that doesn't seem to be it as this happens when I press multiple pads to create new clips in existing tracks

m-m-adams, 2024-09-28:
It definitely is. It reproduces the bug 100% of the time

RJ-Eckie, 2024-11-01:
Easier way to reproduce this:
- press existing clip pad
- press empty clip pad
- release existing clip pad
(1.3.0-DEV-E509B77E)

https://github.com/user-attachments/assets/f273291c-8bc5-473e-826c-f8b542b33426

UI expects the second press to be released first, COPY CLI

Popup stays open when first press is released first instead.
Doing a succesful copy restores normal behavior

commit efdf9350 (2024-11-03): Handle first pad released when two pads are held. Fixes #2693 (#2865)
commit 2c3b74a7 (2024-11-03): Handle first pad released when two pads are held. Fixes #2693 (#2865)
commit e2aae646 (2024-11-03): Merge pull request #2870 from seangoodvibes/Handle-first-pad-released-when-two-pads-are-held

Handle first pad released when two pads are held. Fixes #2693 (#2865)
closed 2024-11-03 by m-m-adams